ouster-lidar/ouster-ros #3: CMake error `file INSTALL cannot find "/root/catkin_ws/src/ouster-ros/../LICENSE".`

Jacob-Flasheye, 2022-10-05:
Hi!

We use `ouster-ros` in our build process, and hadn't pinned a specific commit so our builds stopped working this morning.
After some time, we realized it was because you split this library out of the old repository (good change!), so I've been updating our builds but when I run `catkin_make install`, I get this error message:
```
file INSTALL cannot find "/root/catkin_ws/src/ouster-ros/../LICENSE".
Call Stack (most recent call first):
cmake_install.cmake:136 (include)

Makefile:73: recipe for target 'install' failed
```

I assume that there is some error in the [`CMakeLists.txt`](https://github.com/ouster-lidar/ouster-ros/blob/master/CMakeLists.txt), maybe the `../` should be removed from `../LICENSE`?

Jacob-Flasheye, 2022-10-05:
I pulled the repository locally and can confirm that changing those lines will allow the package to be installed, but I do not know if it is the behaviour you want. If this is a change you want, I can create a PR with the changes.

namjae, 2022-10-05:
this repo was a subdirectory of ouster_example. above files were in the parent directory (that time).

Samahu, 2022-10-05:
@Jacob-Flasheye I believe we overlooked the LICENSE install/copy line. Thanks for reporting the issue, will address this soon.

Jacob-Flasheye, 2022-10-05:
> @Jacob-Flasheye I believe we overlooked the LICENSE install/copy line. Thanks for reporting the issue, will address this soon.

Easy mistake to make, I'm not sure how many ROS users actually uses `catkin_make install`, most guides seem to overlook that feature. Very glad to see such a fast response!

PR #4 (2022-10-05): FSW-4177: update cmake install lines for ouster ros
## Related PRs & Issues
* Closes #3 

## Summary of Changes
* Update `LICENSE` file installation instructions
* Removed the installation of `LICENSE-bin` as deemed unnecessary.
* Update `Dockerfile` to check for installation errors
* Add 4096x5 lidar mode to launch files

## Validation
Use catkin_make command to build and install the project 
```bash
catkin_make
catkin_make install
```
closed 2022-10-05 by Samahu