ouster-lidar/ouster-ros #11: low latency point cloud

shubham-shahh, 2022-10-12:
Hi, @kairenw 
I came across [this](https://github.com/KumarRobotics/ouster_decoder) package by [KumarRobotics](https://github.com/KumarRobotics). 
It has very low latency (<0.2ms) compared to ouster_example (>3ms).
can we incorporate it into this standard driver package

thanks

Samahu, 2022-11-09:
Yes, their implementation converts packets as they arrive rather than waiting for the frame to be ready to start the conversion. It makes total sense but we haven't had the capacity to address it yet.

shubham-shahh, 2022-11-09:
> Yes, their implementation converts packets as they arrive rather than waiting for the frame to be ready/complete to start the conversion. It makes total sense but we haven't had the capacity to address it yet.

any possible disadvantage because of this approach?

Samahu, 2022-11-09:
> > Yes, their implementation converts packets as they arrive rather than waiting for the frame to be ready/complete to start the conversion. It makes total sense but we haven't had the capacity to address it yet.
> 
> any possible disadvantage because of this approach?

Not that I can think of.

The only thing is when I tried their decoder it seemed that their published point cloud was missing certain amount of points when comparing it to the point cloud published by ouster driver. But they have other things in place too. So I can't say the approach itself to blame about missing points.

Samahu, 2022-11-09:
> Not that I can think of.
>
> The only thing is when I tried their decoder it seemed that their published point cloud was missing certain amount of points when comparing it to the point cloud published [...]

I just want to note it has been a while since I have tried their code. So they may no longer have the issue about missing points in their current implementation.