ouster-lidar/ouster-ros #19: Ouster OS1-64-U PTP Synchronization Issue with Ubuntu 20.04

jyang-cpr, 2022-10-25:
I am attempting to use PTP to synchronize the OS1-64-U LiDAR with my Ubuntu 20.04 computer running ROS Noetic. I have been following this documentation (https://static.ouster.dev/sensor-docs/image_route1/image_route2/appendix/ptp-quickstart.html#), and also set `timestamp_mode` to `TIME_FROM_PTP_1588` in the `sensor.launch` ROS launch file; but that has not worked so far.

My system does have a GMR1000 grandmaster clock; however, I have temporarily removed it for simplicity, as I just want synchronization between the LiDAR and my computer for now.

Before I provide details on my `ptp4l`, `phc2sys`, and/or `phc2shm` configuration(s), I want to clarify if I missed an important preliminary step - do I need to configure the LiDAR's `timestamp_mode` to `TIME_FROM_PTP_1588` at the firmware level using TCP first? There seems to be some mention of this detail in https://static.ouster.dev/sensor-docs/image_route1/image_route2/time_sync/time-sync.html#time-synchronization and https://github.com/ouster-lidar/ouster_example/issues/78#issuecomment-516427785. Or, does simply setting `timestamp_mode` to `TIME_FROM_PTP_1588` in `sensor.launch` configure the firmware implicitly already?

Also, if this question is better suited for Ouster customer support, please let me know and I can make a ticket in https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86 instead. Thanks!

Samahu, 2022-10-25:
> Or, does simply setting timestamp_mode to TIME_FROM_PTP_1588 in sensor.launch configure the firmware implicitly already?

It will be configured by the `senosr.launch` if that's what you specified as the value during launch. It will remain on PTP mode until you change it to a different mode (whether through launch file or through the sensor configuration page). 

> Also, if this question is better suited for Ouster customer support, please let me know and I can make a ticket in https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86 instead. Thanks!

If you don't have good handle on PTP configuration, then I would suggest that you contact our customer support. If PTP is not configured properly (i.e. your sensor and host machine are not in sync)., then there is nothing the ouster-ros driver could do about it.

jyang-cpr, 2022-10-25:
Thanks for the quick reply and clarification. I will follow up with customer support.

closed 2022-10-25 by jyang-cpr