ouster-lidar/ouster-ros #22: Two errors in building Dockerfile

mcguiremichael, 2022-11-02:
I tried to build the ouster ros docker container using the following command.

`docker build -t ouster_test .`

There were two error messages, ultimately leading to it to fail to build.

`CMake Error at ouster-ros/CMakeLists.txt:4 (include):
  include could not find load file:

    DefaultBuildType`

`CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:305 (message):
  catkin_package() include dir 'ouster-sdk/ouster_client/include' does not
  exist relative to '/var/lib/build/src/ouster-ros'
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  ouster-ros/CMakeLists.txt:42 (catkin_package)`

How can this be fixed?

Here was the full output.

`Sending build context to Docker daemon  351.7kB
Step 1/18 : ARG ROS_DISTRO=melodic
Step 2/18 : FROM ros:${ROS_DISTRO}-ros-core AS build-env
 ---> 367edf3bdb41
Step 3/18 : ENV DEBIAN_FRONTEND=noninteractive     BUILD_HOME=/var/lib/build     OUSTER_ROS_PATH=/opt/catkin_ws/src/ouster-ros
 ---> Using cache
 ---> a064e66a0432
Step 4/18 : RUN set -xue && [ $ROS_DISTRO = "noetic" ] && PY=python3 || PY=python && echo 'APT::Install-Recommends "0";' > /etc/apt/apt.conf.d/01norecommend && apt-get update && apt-get install -y  build-essential cmake  fakeroot dpkg-dev debhelper  $PY-rosdep $PY-rospkg $PY-bloom
 ---> Using cache
 ---> 93917e1ba8ae
Step 5/18 : RUN apt-get install -y curl libcurl4-openssl-dev
 ---> Using cache
 ---> 7388d042f63f
Step 6/18 : ARG BUILD_UID=1000
 ---> Using cache
 ---> 66b3c9eed365
Step 7/18 : ARG BUILD_GID=${BUILD_UID}
 ---> Using cache
 ---> 8ae466e24401
Step 8/18 : RUN set -xe && groupadd -o -g ${BUILD_GID} build && useradd -o -u ${BUILD_UID} -d ${BUILD_HOME} -rm -s /bin/bash -g build build
 ---> Using cache
 ---> ca256220f449
Step 9/18 : COPY --chown=build:build package.xml $OUSTER_ROS_PATH/package.xml
 ---> Using cache
 ---> f9dc3963396e
Step 10/18 : RUN set -xe         && apt-get update   && rosdep init      && rosdep update --rosdistro=$ROS_DISTRO && rosdep install -y --from-paths $OUSTER_ROS_PATH
 ---> Using cache
 ---> 92ca52eef11f
Step 11/18 : COPY --chown=build:build . $OUSTER_ROS_PATH
 ---> Using cache
 ---> b6bba3f906fd
Step 12/18 : USER build:build
 ---> Using cache
 ---> e77161345da7
Step 13/18 : WORKDIR ${BUILD_HOME}
 ---> Using cache
 ---> 61b9b8928e86
Step 14/18 : RUN set -xe && mkdir src && cp -R $OUSTER_ROS_PATH ./src
 ---> Using cache
 ---> 4270a35a5bd4
Step 15/18 : FROM build-env
 ---> 4270a35a5bd4
Step 16/18 : ENV CXXFLAGS="-Werror -Wno-deprecated-declarations"
 ---> Using cache
 ---> afef28b2e069
Step 17/18 : RUN /opt/ros/$ROS_DISTRO/env.sh catkin_make -DCMAKE_BUILD_TYPE=Release && /opt/ros/$ROS_DISTRO/env.sh catkin_make install
 ---> Running in 92f3ef7b90a7
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /var/lib/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /var/lib/build/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ouster_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ouster_ros'
-- ==> add_subdirectory(ouster-ros)
CMake Error at ouster-ros/CMakeLists.txt:4 (include):
  include could not find load file:

    DefaultBuildType

-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.4
-- Found eigen: /usr/include/eigen3  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- looking for PCL_COMMON
-- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_common.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_common.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libpthread.so  
-- Found CURL: /usr/lib/x86_64-linux-gnu/libcurl.so (found version "7.58.0") 
-- Boost version: 1.65.1
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- ouster_ros: 1 messages, 3 services
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:305 (message):
  catkin_package() include dir 'ouster-sdk/ouster_client/include' does not
  exist relative to '/var/lib/build/src/ouster-ros'
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  ouster-ros/CMakeLists.txt:42 (catkin_package)

-- Configuring incomplete, errors occurred!
See also "/var/lib/build/build/CMakeFiles/CMakeOutput.log".
See also "/var/lib/build/build/CMakeFiles/CMakeError.log".
Base path: /var/lib/build
Source space: /var/lib/build/src
Build space: /var/lib/build/build
Devel space: /var/lib/build/devel
Install space: /var/lib/build/install
Creating symlink "/var/lib/build/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /var/lib/build/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/var/lib/build/devel -DCMAKE_INSTALL_PREFIX=/var/lib/build/install -G Unix Makefiles" in "/var/lib/build/build"
####
Invoking "cmake" failed
The command '/bin/sh -c /opt/ros/$ROS_DISTRO/env.sh catkin_make -DCMAKE_BUILD_TYPE=Release && /opt/ros/$ROS_DISTRO/env.sh catkin_make install' returned a non-zero code: 1`

Samahu, 2022-11-02:
Did you clone ouster-ros with submodule flag as instructed in the [README.md/Getting Started](https://github.com/ouster-lidar/ouster-ros/blob/master/README.md#getting-started) section?

mcguiremichael, 2022-11-03:
Indeed that fixed the problem. Thank you sir.

closed 2022-11-03 by mcguiremichael