ouster-lidar/ouster-ros #23: unwanted points in the point cloud

shubham-shahh, 2022-11-09:
Hi,
In continuation of the issue mentioned [here](https://github.com/ouster-lidar/ouster_example/issues/429) I got in touch with the Ouster team and elaborated this issue as suggested by @kairenw, one of their suggestion was as follows

_Ignore the azimuths/columns of measurement data that are near the shroud column edges. Maybe 2-3 columns on either side as an estimate._

How can I ignore specific azimuths/column in this driver as mentioned in the suggestion

thanks

kairenw, 2022-11-09:
Hi @shubham-shahh,

Even though you can use the [config API](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/client.h#L161) to set the azimuth window (represented as a `std::pair` of ints and provided in millidegrees from 0 to 360000) ignore certain columns, looking at the setup in your ticket with our customer support, I don't suggest this method since the azimuth_window setting must be contiguous but you have several spaced out columns which you'd like to ignore.

My suggestion is to use the PCL PassThrough filter suggested by @Samahu last time. That, or you can fork off `ouster-ros` and change `scan_to_cloud` to give you an unstructured PointCloud as we suggested before.

shubham-shahh, 2022-11-09:
> Hi @shubham-shahh,
>
> Even though you can use the [config API](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/client.h#L161) to set the azimuth window [...]

Hi,
thanks for looking into the ticket, The problem is, even if I use something like a box filter or a pass-through filter based on the measurements of my drone, I see points appearing just outside the filter boundary. and these points appear randomly so increasing the size of filter doesn't help either.

can you please elaborate how unstructured scan can help me? I was able to solve the problem of empty points occurring at origin by using a box filter, but I am not sure how a unstructured scan can help me eliminate these unwanted points occurring near the boundary as mentioned in the ticket, in images and the rosbag

thanks

kairenw, 2022-11-09:
Okay I see. Sorry I didn't read the entire ticket so I thought it was still the issue with the 0 points. Unstructured scan is not going to help you with this.

Here's what I'd try -- let's set the azimuth_window so that it doesn't send points around one of the four columns. If the hypothesis that it's interference between the mounting column and the screen window is correct, then in that quarter of the 360 scan, we shouldn't see the these points.

So what you want to do is find the millidegrees that correspond with the diagram here:
https://static.ouster.dev/sensor-docs/image_route1/image_route2/sensor_operations/sensor-operations.html?highlight=millidegrees#azimuth-window

You can use the C++ or Python Config API or, honestly, just the sensor homepage to configure the azimuth_window to what you need. 

When you get that working, if you could send over a bag of what it looks like, so we're looking at the same results. It might also help to give your sensor's window just a bit of a wipe (:

Samahu, 2022-11-09:
@shubham-shahh are you on the latest of ouster-ros? There was a [bug fix](https://github.com/ouster-lidar/ouster_example/pull/432) related to PCL compilation that may affects the performance of the box filter in case you were using PCL.

shubham-shahh, 2022-11-10:
> @shubham-shahh are you on the latest of ouster-ros? There was a [bug fix back in mid September](https://github.com/ouster-lidar/ouster_example/pull/432) related to PCL compilation that may affects the performance of the box filter in case you were using PCL.

yes, I am using the latest ouster-ros package

Samahu, 2022-11-15:
@shubham-shahh After reviewing the ticket you opened through our support, the issue of unwanted points in the point cloud is due to factors not related to the code itself, applying a box filter won't help either. There could be some software solutions to the problem but I would recommend first working on minimizing any source of interference.

Closing the issue.

closed 2022-11-15 by Samahu