ouster-lidar/ouster-ros #27: PTP time synchronization over USB to Ethernet adapter

Pallav1299, 2022-11-14:
Hi, I am trying to setup PTP time synchronization on [RB5 dev kit](https://developer.qualcomm.com/qualcomm-robotics-rb5-kit). It has a USB3.0 to Ethernet interface, which is being for the LIDAR. I need some help with the following queries:

1.
  - The Software documentation states the following assumption:
[image]

  - The ethernet interface that we have doesn't support Hardware timestamping. As shown here:
[image]
Does it mean we cannot do time synchronization on our development board? What alternatives can we use? 
  - From what I've read, hardware sync is better that software sync in terms of accuracy. Correct me if I'm wrong.

2. To check the performance of time synchronization, I used 

> rostopoic delay /topic_name

utility. This compares the message header timestamp to the current ROS timestamp.
I collected following data:

> delay in **TIME_FROM_PTP_1588** mode over hardware interface: ~0.130
> delay in **TIME_FROM_PTP_1588** mode over software interface: ~0.350
> delay in **TIME_FROM_ROS_TIME** mode: ~0.105

  - How to figure out the amount of delay which suits my application?
  - Is rostopic delay the correct way to compare? If not, what should I use?

3. Extending over 1st question. What's the accuracy loss when we shift to software timestamping supported time synchronization?

Samahu, 2022-11-14:
> * The Software documentation states the following assumption:
> * The ethernet interface that we have doesn't support Hardware timestamping. As shown here:
>   Does it mean we cannot do time synchronization on our development board? What alternatives can we use?

Yes, you cannot use hardware syncronization through a usb-to-ethernet adapter. The alternative is to use hardware with PTP-enabled hardware. Until you have that it is better that you use **TIME_FROM_ROS_TIME** for the _timestamp_mode_, which uses ROS clock for the time but it won't be as accurate as a hardware source.

> * From what I've read, hardware sync is better that software sync in terms of accuracy. Correct me if I'm wrong.

It depends on your application requires, if you use PTP-enabled hardware synchronization you can achieve accuracy of ~ 1 millisecond. Expect much less accuracy with software clocks.

> 2. To check the performance of time synchronization, I used
>
>
> utility. This compares the message header timestamp to the current ROS timestamp. I collected following data:
>
>
> * How to figure out the [...]

I don't know what your application is, so I think you have to answer this questions. But generally speaking if your application has strict time requirements, then you should get a hardware that supports PTP; something is not available  using a a usb-to-ethernet adapater. 

> * Is rostopic delay the correct way to compare? If not, what should I use?

Yes, rostopic is what we use to measure delay. I can not comment on the numbers you posted since you didn't note the topic that you were measuring the delay for. 

> 
> 3. Extending over 1st question. What's the accuracy loss when we shift to software timestamping supported time synchronization?

Unfortunately, we have not characterized this.

Pallav1299, 2022-11-15:
> I don't know what your application is, so I think you have to answer this questions. But generally speaking if your application has strict time requirements, then you should get a hardware that supports PTP; something is not available using a a usb-to-ethernet adapater.

@Samahu I'm working on LIO application.

> delay in **TIME_FROM_PTP_1588** mode over hardware interface: ~0.130
> delay in **TIME_FROM_ROS_TIME** mode: ~0.105

This data refers to the rostopic delay output, when I tried setting up PTP time sync between OS-0 and Hardware timestamping enbaled device. 

> delay in **TIME_FROM_PTP_1588** mode over software interface: ~0.350

While this, I tested on a device which is not capable to do hardware timestamping.
The data is in seconds. Isn't the delay too much, even when hardware timestamping is enabled?

> The alternative is to use hardware with PTP-enabled hardware.

Do you suggest to replace the dev board? Is there any other alternative where we can get good time sync setup which suffices our application with the existing system?

Pallav1299, 2022-11-15:
@Samahu, thanks for your response.

> I don't know what your application is, so I think you have to answer this questions. But generally speaking if your application has strict time requirements, then you should get a hardware that supports PTP; something is not available using a a usb-to-ethernet adapater.

I'm working on LIO application.

> delay in **TIME_FROM_PTP_1588** mode over hardware interface: ~0.130
> delay in **TIME_FROM_ROS_TIME** mode: ~0.105

This data refers to the rostopic delay output, when I tried setting up PTP time sync between OS-0 and Hardware timestamping enbaled device.

> delay in **TIME_FROM_PTP_1588** mode over software interface: ~0.350
While this, I tested on a device which is not capable to do hardware timestamping.
The data is in seconds. Isn't the delay too much, even when hardware timestamping is enabled?

> The alternative is to use hardware with PTP-enabled hardware.

Do you suggest to replace the dev board? Is there any other alternative where we can get good time sync setup which suffices our application with the existing system?

Samahu, 2022-11-15:
>This data refers to the rostopic delay output, when I tried setting up PTP time sync between OS-0 and Hardware timestamping enbaled device.

But which topic you are measuring the delay against, you have `/lidar_packets` and `/points`, if you are talking about `lidar_packets` then the delay should be in the range of 10-20 ms, for `/points` topic it is much more 150-300 ms.

>Do you suggest to replace the dev board? Is there any other alternative where we can get good time sync setup which suffices our application with the existing system?

I do recommend that you have a direct connection to the sensor (no usb-to-ethernet adapter), most NIC adapters have builtin PTP source. Better if you can have a dedicated PTP source.

Samahu, 2022-11-16:
> I've been measuring delay on `/ouster/points` topic. Could you explain what does 150-300ms delay correspond to and how it is related to sensor data rate?
> 

The delay is different from sensor data rate (which is divided into IMU and Lidar data). The data rate is how many samples are you getting per second (aka throughput) while the delay on the other hand measures how long it takes the sample to reach from source to destination (aka latency). 

> As mentioned earlier, the only ethernet interface that is available on our dev board(RB5) is a USB to Ethernet adapter. Are you suggesting to use a network switch with PTP source as master, while LIDAR and dev board as PTP slaves?
>

If this a possible option and having a synchronized data plays an important role of your project.

> Questions:
> 
> 1. Should we be using `**TIME_FROM_ROS_TIME**` mode in our application?

Until you have a PTP clock _properly_ configured on your system you better stick to this mode.

> 2. What other options do we have to set up good time sync in our case?

I think [our tech support](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/-1) may be able to offer better help on this topic

closed 2022-11-17 by Samahu