ouster-lidar/ouster-ros #28: ROS2 multi-casting, multiple processes listening to single LiDAR

mgow001, 2022-11-14:
Hello, 

Could you please provide an example ROS2 launch file for starting Ouster LiDAR in multi-casting mode so multiple processes can listen to the messages? My LiDAR model is OS1-64. 

Thank you!

mgow001, 2022-11-15:
Hello, just writing to follow-up on this. If ROS2 launch example is not available right now maybe you can provide guidelines on how to enable multi-casting for Ouster Lidar so multiple process can listen to the data stream? Thanks!

Samahu, 2022-11-15:
@mgow001 At some point one of our community member submitted a PR against our ouster ROS 1 driver: https://github.com/ouster-lidar/ouster_example/pull/278. It is an old PR but maybe you will find it useful.

mgow001, 2022-11-15:
Thanks @Samahu, I had a look earlier at that PR but I could not make much out of it unfortunately. It would be great to get more detailed multi-casting description if possible. Thanks!

Samahu, 2022-11-15:
There will be some time before we can get to this. But I will keep this issue open.

Samahu, 2023-02-11:
@mgow001 I am working with @ShepelIlya to merge the PR #57, it does work as described but it may benefit some doc which we will get to once we finalize the merge request.

PR #57 (2023-02-20): Feature/multicast ros driver
## Related Issues & PRs
- Original PR: https://github.com/ouster-lidar/ouster_example/pull/278
- ouster-sdk PR: https://github.com/ouster-lidar/ouster_example/pull/480
- Closes #28
- Closes #32

## Summary of Changes
Added multicast data receiver. To use this you need to set udp_dest on ouster configuration page to something like "239.xxx.xxx.xxx" and also set parameter mtp_dest in launch file to IP address of host interface where you want to receive multicast data (for adding this interface to multicast group).

Tested on three Jetson Xavier and two Ouster OS-1 Re [... cut]
closed 2023-02-20 by Samahu