ouster-lidar/ouster-ros #33: Publishing pointclouds in os_sensor frame is misleading

peci1, 2022-11-21:


Aposhian, 2023-05-04:
Is there a reason why the frame on the messages can't be changed to the `lidar_frame`?

Samahu, 2023-05-08:
Unfortunately, I don't have enough background on what or why this was the choice but I'll check on the subject. Especially for the ROS2 driver since it is still in the stage of BETA and we need to get this right.

Aposhian, 2023-05-08:
Based on this documentation, the raw UDP data comes not as XYZ at all, but then can be transformed to the lidar frame, and then transformed to the sensor frame. It will take me some digging in the Ouster SDK to confirm whether it just converts to the lidar frame or also includes the sensor frame.
https://static.ouster.dev/sensor-docs/image_route1/image_route2/sensor_data/sensor-data.html

The pointclouds are stamped with the laser data frame in the unofficial driver: https://github.com/ros-drivers/ros2_ouster_drivers/blob/5425ed023d3c1c5293a9af90d2ae81c5df889334/ros2_ouster/src/ouster_driver.cpp#L127, and by making the clouds be stamped in the lidar frame in my fork of this version, I was able to reproduce results I get with the unofficial driver.

Aposhian, 2023-05-08:
Ok it looks like the ouster sdk (as used by the driver) converts points into the sensor frame, not the data/lidar frame. So I need to figure out why it wasn't working for me...
https://github.com/ouster-lidar/ouster_example/blob/ff46c961b690daae3e18e39999ef8651081cbc66/ouster_client/include/ouster/lidar_scan.h#L354-L355

Aposhian, 2023-05-08:
Ok, I'm pretty sure I got to the bottom of this. The reason for the inconsistency between `ouster_ros` and `ros2_ouster_drivers` is because of a bug in `ros2_ouster_drivers`. `ros2_ouster_drivers` stamps data with the lidar frame, but it is actually in the sensor frame, since the (similar) `make_xyz_lut` function is called with the sensor info, which provides the lidar to sensor transform.
https://github.com/ros-drivers/ros2_ouster_drivers/blob/5425ed023d3c1c5293a9af90d2ae81c5df889334/ros2_ouster/include/ros2_ouster/processors/pointcloud_processor.hpp#L68

So based on this, **I think this ticket can be closed**: I believe this drivers' current behavior is correct. However, it does bring up the following points:
- If the lidar frame is not being used, then should the driver even publish the lidar to sensor transform?
- Maybe in the future it could be a configuration option to allow publishing in the lidar frame (only doing the beam transformation) instead of the sensor frame? I'm not sure if there is demand for that, and if there is, that should be a separate ticket.

peci1, 2023-05-09:
Thanks for the digging.

> **I think this ticket can be closed**

I don't think so. Please, read again my point in the first comment. It is still valid.

> If the lidar frame is not being used, then should the driver even publish the lidar to sensor transform?

In no way. This frame at least allows people to transform the cloud to the frame in which it was actually captured.

> * Maybe in the future it could be a configuration option to allow publishing in the lidar frame

That would currently be the only option for ROS 1. Too much code would get broken if the change was done silently.

@Samahu please, at least get this right for ROS 2 if it's sill in beta.

Samahu, 2023-05-09:
> Ok, I'm pretty sure I got to the bottom of this. The reason for the inconsistency between `ouster_ros` and `ros2_ouster_drivers` is because of a bug in `ros2_ouster_drivers`. `ros2_ouster_drivers` [...]

>I did briefly look into the ros2_ouster_drivers and they are using the lidar_frame
>Maybe in the future it could be a configuration option to allow publishing in the lidar frame
That would currently be the only option for ROS 1. Too much code would get broken if the change was done silently.

@peci1 We are considering to introduce an option for users to select the frame of reference as both of you suggested, the default for ROS1 is to maitain the same behavior, for ROS2 we probably want to set the lidar_frame to be the default since this would make it easier for people who are switching from the community driver to the official ros2 driver.

Aposhian, 2023-05-09:
>for ROS2 we probably want to set the lidar_frame to be the default since this would make it easier for people who are switching from the community driver to the official ros2 driver.

I think the problem is that the community driver _stamps_ the pointclouds with the lidar frame, but it actually _transforms_ them to the sensor frame, so it is lying. You can't make the official driver backward compatible (work with the same TFs) without doing it incorrectly.

It seems that there is demand to get pointclouds in the lidar frame, rather than the sensor frame, so I think it does make sense to make this a configuration option.

Samahu, 2023-06-02:
@peci1 I am reviewing #127 PR that @Aposhian submitted 3 weeks ago, I agree with the conclusions that @Aposhian made. The driver's current xyz_lut object does transform the points into the sensor frame rather than the lidar frame. I think that just simply changing the `frame_id` as the unofficial driver is doing would be incorrect and misleading.

Users who need to have the transform into the lidar frame should rather transform the published point cloud from the sensor frame to the lidar frame using `pcl::transformPointCloud` and utilize the supplied transformation `os_sensor->os_lidar` TF transformation.

Alternatively, we pretty much need to take the same route that @Aposhian made in the linked PR.

peci1, 2023-06-02:
I vote for backporting #127 to ROS 1. That would be the cleanest solution.

If you still can make breaking changes to ROS 2 branch, I suggest publishing in `os_lidar` by default. For ROS 1, this is already lost, the default cannot be changed and most people will still probably blindly use the default output thinking it is in the acquisition frame.

Maybe in ROS 1, a warning could be issued if the driver publishes in `os_sensor`? Something like `transform_to_sensor_frame is set to true. Please note that the projection center of the lidar is not in point (0,0,0), but in (0,0,0.075)`. (or whatever the actual transform is). This might make a few more people aware of this issue. But it would be confusing for those not interested into knowing the projection center.

Samahu, 2023-06-02:
Yeah definitely I have an upcoming upgrade to the ROS2 branch that addresses several issues, one of which is the frame_id of published point cloud, you can take a look at these changes: https://github.com/ouster-lidar/ouster-ros/pull/146 I was about to follow the same approach that currently implemented in Steve's Ouster ROS2 driver for compatibility reasons but yeah we will have to diverge from the Steve's implementation on this particular issue.

Back porting some these improvements to ROS1 will follow after I am done with https://github.com/ouster-lidar/ouster-ros/pull/146.

It makes sense to warn the users for the ROS1 since I want to keep the sensor_frame as the default.

juliangaal, 2023-06-15:
> It makes sense to warn the users for the ROS1 since I want to keep the sensor_frame as the default for ROS1 driver.

I was under the impression that the pointcloud is published already transformed into the sensor frame. Yet, following this discussion, do I understand correctly that the data is published in the sensor frame (*while actually being recorded in the lidar frame*) [here](https://github.com/ouster-lidar/ouster-ros/blob/3f01e1d7001d8d21ac984566d17505b98905fa86/src/os_cloud_nodelet.cpp#L187) and the transform is published separately?

Samahu, 2023-06-15:
> I was under the impression that the pointcloud is published already transformed into the sensor frame. Yet, following this discussion, do I understand correctly that the data is published in the [...]

No, in ROS1 the data is published in the sensor frame and the `frame_id` of the PointCloud2 message is correctly set to the sensor frame. However, some of our users who need to do ray tracing would rather have the PointCloud2 message published in the lidar frame, which isn't readily available. Simply changing the name of the `frame_id` to the lidar frame would not be correct. So if you expect the PointCloud to be published in the sensor frame then that's what is currently take place in ROS1 and it is correct. If you require the PointCloud be published in the lidar frame however you would need to wait until I back port parts of #146 to ROS1 that give you the option to do so. The default behavior for the ROS1 driver would be to keep publishing the point cloud in the sensor frame (with a warning or a hint). For ROS2  though the default would be to publish the PointCloud in the lidar frame.

juliangaal, 2023-06-15:
> Simply changing the name of the frame_id to the lidar frame would not be correct.

Because the data was already transformed into `sensor_frame`? At which level does this happen: the driver or the sdk? I can't  seem to find the relevant code in the cloud nodelet

Samahu, 2023-06-15:
> > Simply changing the name of the frame_id to the lidar frame would not be correct.
> 
> Because the data was already transformed into `sensor_frame`? At which level does this happen: the driver or the sdk? I can't seem to find the relevant code in the cloud nodelet

The object returned by the non-parameterized `make_xyz_lut` method generates a transformation that projects the xyz coordinates into the sensor frame. The mainline of ros1 & ros2 drivers currently uses this method. This is changing with #146 and #149 for ros2. I will propagates some of these changes to ros1 too.

PR #146 (2023-06-23): ROS2 compatibility mode
## Related Issues & PRs
- foxy's #149
- resolves #8
- resolves #50
- resolves #33
- resolves #63
- partially addresses #69
- related #76
- related #110
- incorporates #127
- related #137
- related #144

## Summary of Changes
- Build compatibility mode with [Steve Macenski's implementation of Ouster driver for ROS2](https://github.com/ros-drivers/ros2_ouster_drivers)
- Combine the functionality of **os_sensor** and **os_cloud** into a single node/component named **os_driver**
- Retain the **os_sensor** and **os_cloud** for backward compatibility and also still be able to [... cut]
PR #170 (2023-07-25): SW-5148: back port improvements in ros 2 to ros 1
## Related Issues & PRs
- original PR: #157 
- related: #146 
- related: #149
- related: #156 
- related: #162
- related: #163 
- resolves: #8 
- resolves: #33 
- resolves: #65 for custom frame names.
- resolves: #76
- resolves: #105 and #9 by implemented a switch between static and dynamic broadacasts of TF messages. 
- resolves: #110
- resolves: #137
- resolves: #144
- resolves:  #155
- incorporates: #127
- resolves: #144
- resolves: #147

## Summary of Changes
- Added the **ThreadSafeRingBuffer**
  - with minimal unit tests for it
- Switch t [... cut]
closed 2023-07-25 by Samahu