ouster-lidar/ouster-ros #35: How to change /os_cloud_node/points time stamp to ros time?

JACKLiuDay, 2022-11-30:
Hi, thank you for your work with ouster-ros. 
I am using ouster OS-0-64 lidar to run my LIO algorithm. And I choose to use imu 3dm-gx5-25 instead of ouster imu. But when I run ouster.launch. The topic /os_cloud_node/points time stamp is not ros time. While my imu data time stamp is ros time. So I need to sync my lidar data and imu data by ros time. But I don't know how to get the lidar topic data with ros time. 
Could you give me some advice about how to change the driver file. So I can get the points data with ros time.
Thank you.

Samahu, 2022-11-30:
Hi @Dayu,

you should be able to achieve so by passing the value “TIME_FROM_ROS_TIME”
to the *timestamp_mode* argument when running the ‘sensor.launch’. This
will set the timestamp of the IMU and PointCloud messages using ROS time
instead of the sensor timestamp.

Ussama

JACKLiuDay, 2022-11-30:
> Hi @dayu, you should be able to achieve so by passing the value “TIME_FROM_ROS_TIME” to the *timestamp_mode* argument when running the ‘sensor.launch’. This will set the timestamp of the IMU and PointCloud messages using ROS time instead of the sensor timestamp. Ussama
> -- *Ussama Naal* San Francisco, CA <http://ouster.com/>

Hi,thank you for your quick reply. I just found my ouster-ros is old version and the launch file in it does not have TIME_FROM_ROS_TIME node. I will try the new version ouster-ros driver and give your a test result. Thanks again.

Samahu, 2022-12-02:
I haven't header back from you but I hope that I have answered your question, closing the issue.

closed 2022-12-02 by Samahu

JACKLiuDay, 2022-12-03:
> I haven't heard back from you but I hope that I have answered your question, closing the issue.
[image]

I just installed the new ouster driver. And I built my map successfully. Thank you for your advice.