ouster-lidar/ouster-ros #36: Failed when trying to visualize point cloud in RVIZ

andressaff, 2022-12-01:


Samahu, 2022-12-01:
Hi @andressaff, this is interesting. I can see that you are connected to the LIDAR but you are not getting the reflect image, It is not clear but it seems that you don't have the point cloud too?

After you launch the sensor, can you list running node(lets) by executing the command in a separate terminal?
`rosnode list`

Also can you check if these topics have are active?
```bash
rostopic echo /ouster/imu_packets
rostopic echo /ouster/lidar_packets
rostopic echo /ouster/points
```

andressaff, 2022-12-01:
@Samahu I can check if the nodes are active, I even used the commando below `rqt_graph` and got this image
[image]

Samahu, 2022-12-01:
So I haven't recently tested the driver with ROS melodic, I would have to try to see if this has anything to do with the issue.

Meanwhile, to eliminate other factors at play, can I ask you to clone the ouster-ros repository into a separate catkin workspace (so only ouster-ros driver is running) and check if the issue still present?

andressaff, 2022-12-02:
@Samahu The ros ouster is already in a different catkin workspace than the FAST LIO, and I also ran the command `roslaunch ouster_ros sensor.launch sensor_hostname:=os-122137000149.local` just to view the point cloud of the ouster sensor

Samahu, 2022-12-05:
@andressaff So I have tested the current ouster-ros driver under **ROS melodic** and was able to get the point cloud presented in RVIZ including reflectivity and other sensor data layers. My recommendation is that you make sure you have all correctly clone the project as is documented in [the project readme](https://github.com/ouster-lidar/ouster-ros#getting-started) and have installed all the required dependencies by running **rosdep** within your catkin workspace:
```bash
rosdep install -y --from-paths src/ouster-ros
```

If you are still having trouble I would recommend contacting our [customer support](https://ouster.atlassian.net/servicedesk/customer/portal/8/).

Hope this helps

Ussama

andressaff, 2022-12-07:
@Samahu I managed to solve the problem, it was a problem with the network interfaces, I used the command `sudo /etc/init.d/networking restart` and it worked again. Thank you!!!!

andressaff, 2022-12-07:
@Samahu I managed to solve the problem, it was a problem with the network interfaces, I used the command `sudo /etc/init.d/networking restart `and it worked again. Thank you!!!!

closed 2022-12-07 by andressaff