ouster-lidar/ouster-ros #37: How to rotate the lidar orientation frame?

bclaw524, 2022-12-07:
Hello,

I have an ouster (OS0) in a fixed position, I would like to rotate the orientation, so the X-axis is pointing 90 degrees to the left (like below)

[image]

I am a bit confused by the manual on how to do so.

Do I just edit the data_test.json file from

`{
"lidar_to_sensor_transform": [-1, 0, 0, 0, 0, -1, 0, 0, 0, 0, 1, 36.180, 0, 0, 0, 1]
}`

to 

`{
"lidar_to_sensor_transform": [0, -1, 0, 0, 1, 0 , 0, 0, 0, 0, 1, 36.180, 0, 0, 0, 1]
}`

Would this be correct?

Samahu, 2022-12-08:
Hi @bclaw524,

It depends on what you are trying to achieve:
* If this is only applies to test data then editing the json file should be fine
* If this applies to a live sensor you would have to apply a rotation to the sensor `lidar_to_sensor_transform` such that the resulting frame is rotated. If this is within ROS and the sensor is part of a larger composition then you can simply apply the rotation at through the parent transform.

bclaw524, 2022-12-09:
It's closer to the latter.

I'm trying to calibrate cameras to the lidar so their point clouds would align properly in rviz. 

The calibration package I'm using does the calculations best when the lidar is oriented in the same direction the camera is facing (so both the camera and lidar  "front" axis are parallel to one another). 

I have 4 cameras pointing in different directions, with the lidar being in the middle (setup similar to the simulation below).  I would like to "rotate" the orientation of the ouster without having to physically rotate the lidar.  

[image]

> If this is within ROS and the sensor is part of a larger composition then you can simply apply the rotation at through the parent transform.

How do I do this exactly?

Samahu, 2022-12-09:
@bclaw524 try adding something like the following to one of the sensor launch file:

```xml
  <node pkg="tf" type="static_transform_publisher" name="os_parent" args="0 0 0 -1.57 0 0 os_parent os_sensor 10" />
```

This would create a parent link name **os_parent** to the sensor main link **os_sensor** but with a rotation of 90 degrees CCW. Then you could use this transform when positioning the sensor in your scene or setup.

Hope this helps
Ussama

Samahu, 2022-12-12:
@bclaw524 I haven't heard back from you, I hope my answer helped you with the issue you are dealing with. closing the issue.

closed 2022-12-12 by Samahu

bclaw524, 2022-12-13:
@Samahu sorry for the late reply, was still in the testing phase earlier today. 

 managed to "solve it" by rotating the pointcloud data directly. 

Creating a parent link didn't really work as when I rotated the lidar in rviz, the pointcloud data that was transmitted by the sensor would still rotate with the frame. 

Thanks for trying though.