ouster-lidar/ouster-ros #44: Bad data output on RaspberryPi

MatCapLynx, 2023-01-05:
Hi, 
I'm using a RPi 4B 8Gb to receive data from an OS1-32. For the first few seconds I receive the whole pointcloud but then there is some black strips and I only receive the closest points on the pointcloud.
[image]

I've noticed that this depends of the lidar mode. It seems to depend of the number of points received. In 512x10 everything is fine but this problem happens when using 1024x10, 1024x20, 2048x10.
May be this is coming from the processing power, this is the %CPU it got with top command:
* 1024x10: 195%CPU
* 2048x10: 220%CPU

I've checked with my computer instead of the RPi and everything is fine in all lidar modes.

Samahu, 2023-01-10:
Hi @MatCapLynx,

I haven't tried to run the driver on a RPi 4 myself but I plan to.. I have observed incomplete point clouds when trying to run the ouster-ros driver on edge devices similar to RPi in capability. Where the driver runs fine with lidar modes 512x10, 512x20, and even 1024x10 seemed okay to me but anything higher than that such as 1024x20 or 2048x10 had poor performance. It is on my to do list to dig deep into the issue and figure out why this is the case and how we can resolve or mitigate or resolve this behavior. I will keep this ticket open and keep you updated.

Grisly00, 2023-01-19:
Hi,
we experienced similar issues on the Raspi. The solution for us at the moment is to record the lidar_packets directly instead of Pointcloud2. Later you can replay the bagfile on a more powerful computer and transform the recorded lidar_packets into Pointcloud2. 
However, I would also be happy to hear about any improvements that would make this step unnecessary.
Thanks!

MatCapLynx, 2023-02-17:
I've tried again in 2048x10 mode and it seems much better. I believe it is because I built the package with the `-DCMAKE_BUILD_TYPE=Release` option. I don't really remember if I used it before.
I also used an older Ouster so it's probably running a different firmware. I don't know if it changes anything.

The ouster nodelet now uses ~120%CPU.

However there is still some parts of the pointcloud disappearing for an instant but it's not really a problem. The pointcloud is much more usable than before.
[image]

Samahu, 2023-02-17:
There was an update to the drive back in December https://github.com/ouster-lidar/ouster-ros/pull/40 that improved its performance and the stability, did you use the updated driver with your test? Just curious if the CPU utilization that you reported (120%) included the update. I still haven't got the time to test with RPi :tired_face:

Samahu, 2023-11-13:
@MatCapLynx we had implemented several fixes towards the ROS1 driver since #249 that should reduce dropped packets and reduce overall CPU utilization, could you give the last update a try and let us know if could close this issue?

Samahu, 2023-11-20:
@MatCapLynx I haven't heard back from you. I am closing the issue, if you are (or someone else) still having problems then they can respond to this or open new ticket.