ouster-lidar/ouster-ros #47: Scans with missing packets at the beginning have wrong timestamp

peci1, 2023-01-13:
https://github.com/ouster-lidar/ouster-ros/blob/b5393ca9ceb12823ddc78828c0326f0d32af1c51/src/os_cloud_nodelet.cpp#L139-L143

This code takes the first valid timestamp in the scan as the timestamp of the whole scan. However, the usual understanding of `header.stamp` field is when the scan acquisition began. I understand that when the first packets are missing, this information is not exactly available. However, given the regularity and precision of the rotation motor in the lidar, the timestamp of column 0 could be very easily extrapolated. That would seem to me as a better solution.

Samahu, 2023-01-13:
That's a good point @peci1, I didn't know about the convention/understanding that `header.stamp` represents the beginning of acquisition. I think we can easily extrapolate the time based on packet/column frequency.

nkhedekar, 2023-01-25:
Just to add here, most LiDAR based SLAM methods make this assumption so this enhancement would definitely be very appreciated :)

Samahu, 2023-01-25:
@nkhedekar thanks for feedback. we will prioritize this and have a fix soon.

ShepelIlya, 2023-01-30:
+1 to this issue - we have same troubles with our SLAM algos. So, then CPU utilization is high on our jetson agx xavier where works our slam and ouster driver approximately 1 time per minute ouster ROS driver produces cloud with timestamp 2-5 secs in future, so our SLAM is crushes.

Samahu, 2023-02-18:
Just as a quick update: I do have a preliminary solution to this issue, expect a PR next week!

Samahu, 2023-02-23:
> Just as a quick update: I do have a preliminary solution to this issue, expect a PR next week!

@peci1 and all others who are affected, I have pushed a fix to issue [here](https://github.com/ouster-lidar/ouster-ros/pull/67) with a brief description of the technique used to estimate the beginning of scan_ts when the lidar scan is missing timestamp info at the beginning of the frame. Could please take a look and give it a try and see if it addresses the issue you are facing.
There is multiple techniques that one could use to estimate the missing value but I found out that using linear interpolation between the two closest values would yield the best estimate of the missing value. Your feedback is welcome.
  
Note that this solution does not /fully/ address the case when `TIME_FROM_ROS_TIME` is used.

ShepelIlya, 2023-02-27:
Looks like working solution for [our setup](https://github.com/ouster-lidar/ouster-ros/pull/67#issuecomment-1446464170) (btw i have tested it in multicast mode). 
Waiting for merge in master branch.

PR #67 (2023-03-07): Implement a technique to estimate scan_ts when packets are missing at beginning of a lidar scan
## Related Issues & PRs
Closes #47 
Related #46 

## Summary of Changes
* Implement a technique to estimate `frame_ts` and `scan_ts` when packets are missing at beginning of a LIDAR scan
* The technique utilizes linear interpolation to impute the value of the frame time
  - The timestamp of the very first frame is extrapolated based on column time spacing
[image]

> **Note**
> The `TIME_FROM_ROS_TIME` case extrapolates frame time based on the reception time of the first packet/column to arrive of a  [... cut]
closed 2023-03-07 by Samahu