ouster-lidar/ouster-ros #48: No Topic information of Ros2

jeychandar, 2023-01-18:
Hi. I tried converting pcap to bag format. I tried generatating pointcloud information but there is no topic information for ros2. It shows error like this . Could you please help me out .Thanks in advance
[image]

Samahu, 2023-01-18:
Was the `my.bag` file recorded using ouster_ros driver (ros1) or was it recorded using a different driver? What happens if you use `rosbag info` on the file?

jeychandar, 2023-01-19:
Hi. I converted pcap to bag using this tool https://github.com/Krishtof-Korda/pcap-to-bag.git. After that I tried to play that error happens. Rosbag info topic information it is showing for ros2 it is not. And I tried to pass the recorded data into ouster_ros master replay it was successfully ran and generated bag file with current timestamp.bag format but still there is no ros2 topic information for that too

Samahu, 2023-01-19:
I think the tool you referenced uses ROS1 when performing the conversion. You would need to use ros1 bag utilities to replay the files (i.e you would need to use the command `rosbag info` and not `ros2 bag info`). Otherwise, if you still want to use ROS2 utilities to replay the generated bag file then you can try to convert it to ros2 compatible format using the following tool: https://ternaris.gitlab.io/rosbags/topics/convert.html

jeychandar, 2023-01-24:
Hi I have tried as you said but it converts ros2 .db3 format. I want to use rosbag_v2 storage id that requires .bag format not sqlite3 storage id. https://github.com/TixiaoShan/LIO-SAM.git. In this repo the sample data works both on rosbag and ros2 .bag format. I used that sample data and run with this repo https://github.com/rsasaki0109/li_slam_ros2.git. So far everything works fine sir. Could you help me out. This will be helpful for me or could you suggest me how to play pcap format files directly.

Samahu, 2023-01-26:
Unfortunately, I need to look in depth into this and I am busy building the ros2 driver. So I will be at least another week before I can look into the case. What I suggest is that could try to replay/stream the pcap file using some tools like [tcpreplay](https://tcpreplay.appneta.com/) or [udpreplay](https://github.com/rigtorp/udpreplay) and then record the bag file directly from it using your ros toolset and see if you get the proper output.

Samahu, 2024-08-15:
Hi @jeychandar, you can utilize any of the two utility scripts supplied in this PR to convert between our ROS1/ROS2 bag file formats (they are not production ready yet but with little changes you can achieve the conversion you like).

closed 2024-08-15 by Samahu

Samahu, 2024-08-15:
I forgot to link the PR my original post :facepalm: .. appolgies. Here it is https://github.com/ouster-lidar/ouster-ros-extras/pull/2