ouster-lidar/ouster-ros #50: Plans for ROS2?

Grisly00, 2023-01-19:
Hi everyone,

I would like to revisit this issue [ouster-lidar/ouster_example#205](https://github.com/ouster-lidar/ouster_example/issues/205) from two years ago to get some information on you plans about ROS2.
More and more applications and drivers are shifting to ROS2 which makes in increasingly frustrating to work with ROS1 packages.

I am aware that there is a ROS2 driver by the community ([https://github.com/ros-drivers/ros2_ouster_drivers](https://github.com/ros-drivers/ros2_ouster_drivers)), but it still lacks some key features such as the Dual Return mode.

From Ouster distribution we got the information, that a official ROS2 driver is in the works and could potentially be released in the first half o 2023. Here on github it more seems like you try to cooperate with the community driver.

Could elaborate a bit on what to expect from the upcoming months as a user. It is very time consuming to implement things like ROS bridge or switching Distros, so this could help us a lot qith planing on where to put resources.

Thanks in advance!

Samahu, 2023-01-19:
@Grisly00, I understand your frustration. We should have an early prototype implementation of ROS2 driver this month.

Grisly00, 2023-01-22:
Happy to hear you are working on that!
We will be looking forward to it, thank you for your fast answer!

mvancleaver, 2023-01-26:
+1 to this.

Our team is migrating to ROS2 and an official driver would be great

Samahu, 2023-01-27:
@mvancleaver Thanks for the feedback, keep an eye #51

DrShIkIgAmy, 2023-02-03:
I and my team has already implemented ros2 pkg for ouster driver. And to keep sync with official repo, we could gladly put it in corresponding branch of ouster-ros repo. @Samahu, what do you think about that?

ShepelIlya, 2023-02-03:
> My team and I has already implemented ros2 pkg for ouster driver. And to keep sync with official repo, we could gladly put it in corresponding branch of ouster-ros repo. As far as I know, my colleague @ShepelIlya has already offered it. So @Samahu, what do you think about that?

Yeeah +1 to this. We are ready to share our code for ROS2. It have been tested already on our robot with 2 OS1-32.

Samahu, 2023-02-03:
@DrShIkIgAmy @ShepelIlya thanks so much for offering the support on this, and just to let you know I am always in favor of accepting any beneficial or valuable contributions, however, we do have certain requirements of the ros2 diver from multiple customers/users that we need to fulfill.

Good news, I should be wrapping up the #51 PR today and I expect it to be merged early next week once reviews are done.

Samahu, 2023-02-13:
Just as an update, the linked PR is still in review, I expect that it would be merged this week.

hodnajit, 2023-02-16:
> Just as an update, the linked PR is still in review, I expect it would be merged this week.

Oooh looking forward to try that one :)

Samahu, 2023-02-21:
We have completed our of merge of #51 PR, this is the minimal viable product and the driver is released as a beta version. More work and improvements is planned towards the ros2 driver which we will deliver over the next month. 
Please give it a try when you have the time:
- For ROS2 _rolling_ and _humble_ distros: https://github.com/ouster-lidar/ouster-ros/tree/ros2
- For ROS2 _foxy_ distro: https://github.com/ouster-lidar/ouster-ros/tree/ros2-foxy

PR #51 (2023-02-20): ROS2 driver MVP
## Related Issues & PRs
- Related #52
- Related #58
- Closes #50

## Summary of Changes
- Porting ROS1 driver to ROS2
  - Retained previous functionality
- Utilization of node/component composition of ros2
- Applied `SensorDataQoS` as the QoS policy for all sensor data (hardcoded).
- Partial implementation of ROS2 Lifecycle management for the **os_sensor** node
  - Added a new service to the **os_sensor** node named `reset`
  - The ROS2 Lifecycle management implementation is partial since it does not support the change of certain sensor settings
  - When non legacy lidar profil [... cut]
PR #146 (2023-06-23): ROS2 compatibility mode
## Related Issues & PRs
- foxy's #149
- resolves #8
- resolves #50
- resolves #33
- resolves #63
- partially addresses #69
- related #76
- related #110
- incorporates #127
- related #137
- related #144

## Summary of Changes
- Build compatibility mode with [Steve Macenski's implementation of Ouster driver for ROS2](https://github.com/ros-drivers/ros2_ouster_drivers)
- Combine the functionality of **os_sensor** and **os_cloud** into a single node/component named **os_driver**
- Retain the **os_sensor** and **os_cloud** for backward compatibility and also still be able to [... cut]
closed 2023-02-21 by Samahu