ouster-lidar/ouster-ros #54: Error setting config:  CurlClient::execute_get failed for the url - Couldn't connect to server

SvenMuc, 2023-01-27:
I'm trying to connect to an Ouster OS1-128 (firmware v2.4.0) with the ouster_ros node.
With the Ouster Studio I can connect correctly to the sensor and visualize the point clouds correctly.

The Ouster-OS1-128 is configured with an fixed IP.

When I rune the following command on the same machine, the communication fails.
Any idea?

**Ouster-Ros Node**
`roslaunch ouster_ros sensor.launch sensor_hostname:=192.0.2.100 udp_dest:=192.0.2.10 lidar_port:=7502 imu_port:=7503 viz:=false`

**Log**
```
SUMMARY
========

PARAMETERS
 * /ouster/os_cloud_node/tf_prefix: 
 * /ouster/os_cloud_node/timestamp_mode: 
 * /ouster/os_node/imu_port: 7503
 * /ouster/os_node/lidar_mode: 
 * /ouster/os_node/lidar_port: 7502
 * /ouster/os_node/metadata: 
 * /ouster/os_node/sensor_hostname: 192.0.2.100
 * /ouster/os_node/timestamp_mode: 
 * /ouster/os_node/udp_dest: 192.0.2.10
 * /ouster/os_node/udp_profile_lidar: 
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)
  /ouster/
    img_node (nodelet/nodelet)
    os_cloud_node (nodelet/nodelet)
    os_node (nodelet/nodelet)
    os_nodelet_mgr (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[ouster/os_nodelet_mgr-1]: started with pid [5789]
process[ouster/os_node-2]: started with pid [5791]
process[ouster/os_cloud_node-3]: started with pid [5793]
process[ouster/img_node-4]: started with pid [5795]
process[relay_lidar-5]: started with pid [5797]
process[relay_imu-6]: started with pid [5798]
[ INFO] [1674827893.786712145]: Initializing nodelet with 8 worker threads.
[ INFO] [1674827894.766520780]: Loading nodelet /ouster/os_node of type nodelets_os/OusterSensor to manager os_nodelet_mgr with the following remappings:
[ INFO] [1674827894.797790246]: Will send UDP data to 192.0.2.10
[ERROR] [1674827894.799617669]: Error setting config:  CurlClient::execute_get failed for the url: [http://192.0.2.100/api/v1/system/firmware] with the error message: Couldn't connect to server
[FATAL] [1674827894.799853300]: Failed to load nodelet '/ouster/os_node` of type `nodelets_os/OusterSensor` to manager `os_nodelet_mgr'
================================================================================REQUIRED process [ouster/os_node-2] has died!
process has died [pid 5791, exit code 255, cmd bash -c sleep 3; $0 $@ /opt/ros/noetic/lib/nodelet/nodelet load nodelets_os/OusterSensor os_nodelet_mgr __name:=os_node __log:=/home/ae/.ros/log/3344174c-9e42-11ed-84fd-553d1d5567c9/ouster-os_node-2.log].
log file: /home/ae/.ros/log/3344174c-9e42-11ed-84fd-553d1d5567c9/ouster-os_node-2*.log
Initiating shutdown!
================================================================================
[relay_imu-6] killing on exit
[relay_lidar-5] killing on exit
[ouster/img_node-4] killing on exit
[ouster/os_cloud_node-3] killing on exit
[ouster/os_node-2] killing on exit
[ouster/os_nodelet_mgr-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Samahu, 2023-01-27:
The log states that: `failed for the url: [http://192.0.2.100/api/v1/system/firmware] with the error message: Couldn't connect to server` which implies that the device is not reachable at the address `192.0.2.100`

May I ask you what firmware version is installed on the sensors?

Samahu, 2023-01-27:
* Are you actually able to ping the IP: `192.0.2.100` from where you run the client?

* What happens if copy paste the URL: `http://192.0.2.100/api/v1/system/firmware` into your browser?

SvenMuc, 2023-01-28:
> * Are you actually able to ping the IP: `192.0.2.100` from where you run the client?

Yes I can ping the IP 192.0.2.100 and my own one 192.0.2.10. I can also configure the sensor vie the browser via the IP address. Even Ouster Studio works correctly on the same machine.

> * What happens if copy paste the URL: `http://192.0.2.100/api/v1/system/firmware` into your browser?

I will check this, when I'm back to office in Monday.

SvenMuc, 2023-01-30:
We fixed the issue. It was a misconfiguration of the proxy settings.

closed 2023-01-30 by SvenMuc

Zdctryto, 2023-03-31:
I have encountered the same problem. Can you tell me how to solve it? Thank you.

jando46, 2023-04-03:
> We fixed the issue. It was a misconfiguration of the proxy settings.

Hi Sven, I have the same issue trying to launch the sensor.launch file. Could you suggest me how to solve by editing proxy settings? Thank you

zillur-av, 2023-06-16:
Hi @SvenMuc  I am getting the same error in ros2 even after unsetting those variables. My ping works too. `udp_dest, lidar_port, imu_port`, are they random? Or how did you select them?

SvenMuc, 2023-06-17:
@zillur-av In our case the issue was caused by the proxy settings. We use fixed IP addresses and therefore we set `udp_des` to the IP of the receiving ethernet port. Otherwise you won't get any point clouds. `lidar_port` and `imu_port` are chosen randomly, if you don't set them to a fixed value. In our case we use fixed values for those attributes as well.

zillur-av, 2023-06-17:
> @zillur-av In our case the issue was caused by the proxy settings. We use fixed IP addresses and therefore we set `udp_des` to the IP of the receiving ethernet port. Otherwise you won't get any [...]

I resolved that issue by using sensor serial number as hostname, still getting no messages. See the issue #151. Perhaps something to do with the `udp_des` or `time stamp mode`. After connecting the sensor to the computer, I setup `link local` ip address. From `ouster studio` software, I am getting this `udp_des` ip address.