ouster-lidar/ouster-ros #56: unable to launch sensor.launch

AKSHAYRAMPP, 2023-01-28:
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/nvidia/ouster_ws/devel/lib/libouster_ros.so: undefined symbol: _ZNK6ouster9LidarScan5fieldIjLj0EEEN5Eigen3RefIKNS2_5ArrayIT_Lin1ELin1ELi1ELin1ELin1EEELi0ENS2_8internal11conditionalIXsrS7_21IsVectorAtCompileTimeENS2_11InnerStrideILi1EEENS2_11OuterStrideILin1EEEE4typeEEENS_6sensor9ChanFieldE
[FATAL] [1674891721.774745183]: Failed to load nodelet '/ouster/os_cloud_node` of type `nodelets_os/OusterCloud` to manager `os_nodelet_mgr'
[FATAL] [1674891721.775106561]: Failed to load nodelet '/ouster/img_node` of type `nodelets_os/OusterImage` to manager `os_nodelet_mgr'
================================================================================REQUIRED process [ouster/os_nodelet_mgr-2] has died!
process has died [pid 2000, exit code 127, cmd bash -c sleep 2; $0 $@ /opt/ros/melodic/lib/nodelet/nodelet manager __name:=os_nodelet_mgr __log:=/home/nvidia/.ros/log/338c96dc-9edf-11ed-8353-00044bc4960c/ouster-os_nodelet_mgr-2.log].
log file: /home/nvidia/.ros/log/338c96dc-9edf-11ed-8353-00044bc4960c/ouster-os_nodelet_mgr-2*.log
Initiating shutdown!
================================================================================

board:nvidia jetson tx2
os:ubuntu 18.04
ros melodic

Samahu, 2023-01-30:
The line `_ZNK6ouster9LidarScan5fieldIjLj0EEEN5Eigen3RefIKNS2_5ArrayIT_Lin1ELin1ELi1ELin1ELin1EEELi0ENS2_8internal11conditionalIXsrS7_21IsVectorAtCompileTimeENS2_11InnerStrideILi1EEENS2_11OuterStrideILin1EEEE4typeEEENS_6sensor9ChanFieldE` suggests that you may have issues with eigen library that you have installed on your system. Can you report Eigen's library version number through `dpkg -l | grep libeigen3-dev`

Try also to perform a clean rebuild .. maybe that could surface the issue at compile time.

Samahu, 2023-02-02:
closing the issue due to inactivity.

closed 2023-02-02 by Samahu