ouster-lidar/ouster-ros #59: Build issue

santumuchetty, 2023-02-16:
Build fails with the below error
```
src/ouster-ros/ouster-sdk/ouster_client/src/image_processing.cpp:145:31:   required from here
/usr/local/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:23:29: error: invalid use of incomplete type 'struct Eigen::internal::traits<double>'
   23 | template<typename T> struct traits<const T> : traits<T> {};
      |                             ^~~~~~~~~~~~~~~
/usr/local/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:17:29: note: declaration of 'struct Eigen::internal::traits<double>'
   17 | template<typename T> struct traits;
```

looks like this is because of the below functions from src/ouster-ros/ouster-sdk/ouster_client/src/image_processing.cpp line #145

```
 // use overloads vs templates so implicit conversion to Eigen::Ref still works
void AutoExposure::operator()(Eigen::Ref<img_t<float>> image,
                              bool update_state) {
    update(image, update_state);
}

void AutoExposure::operator()(Eigen::Ref<img_t<double>> image,
                              bool update_state) {
    update(image, update_state);
}
```

The build is fine by commenting out the first function.

Eigen version : 3.3.7
gcc version: 9.4.0
Ubuntu 20.04

Samahu, 2023-02-16:
Can I ask if you are building the **ouster-ros driver** in a standalone project or did you encounter this error when integrating the driver into a larger project? I ask this because the larger projects could affect the selected Eigen package (maybe you utilize vcpkg along with ROS).

Also mind checking the version of Eigen library that is actually in use when compiling the project by embedding the following snippet somewhere in the driver code:

```cpp
#pragma message("eigen world: " BOOST_PP_STRINGIZE(EIGEN_WORLD_VERSION))
#pragma message("eigen major: " BOOST_PP_STRINGIZE(EIGEN_MAJOR_VERSION))
#pragma message("eigen minor: " BOOST_PP_STRINGIZE(EIGEN_MINOR_VERSION))
``` 

Please try that and let me know what you get.

santumuchetty, 2023-02-16:
```
-- Found Eigen: /usr/include/eigen3 (Required is at least version "3.1") 
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- Found Boost: /usr/include (found suitable version "1.71.0", minimum required is "1.55.0") found components: system filesystem date_time iostreams regex 
```
It is a standalone project. I just cloned and built the project.

Samahu, 2023-02-17:
Googling around I found other instances similar to the issue you are dealing with, check this: https://github.com/cdcseacave/openMVS/issues/164

I couldn't generate the problem on my machine even after switch my compiler to gcc-9.4.0. Is upgrading to a newer version of gcc an option to you?

Alternatively, you can leave the line commented out on your machine but since none of our users has reported this problem .. I am not leaning into removing the function definition.

santumuchetty, 2023-02-17:
```
src/ouster-ros/src/os_ros.cpp:26:72: note: #pragma message: eigen world: 3
   26 | #pragma message("eigen world: " BOOST_PP_STRINGIZE(EIGEN_WORLD_VERSION))
      |                                                                        ^
src/ouster-ros/src/os_ros.cpp:27:72: note: #pragma message: eigen major: 3
   27 | #pragma message("eigen major: " BOOST_PP_STRINGIZE(EIGEN_MAJOR_VERSION))
      |                                                                        ^
src/ouster-ros/src/os_ros.cpp:28:72: note: #pragma message: eigen minor: 7
   28 | #pragma message("eigen minor: " BOOST_PP_STRINGIZE(EIGEN_MINOR_VERSION))
      |                                                                        ^

```
Here's the output of eigen version.

Samahu, 2023-02-17:
I have the same version as you and I am using Ubuntu 20.04 but I was using gcc-11, after downgrading to gcc-9.4.0 which exactly what you have I still can't produce the issue. So I don't think there is something particularity bad about our code. But I would be interested to know what's causing it to fail on your machine.

Samahu, 2023-03-02:
Not sure how to deal with this issue since it seems that it only affects your installation I will close it as is. If this affects more people only then I would consider commenting out the `traists<double>` line.

closed 2023-03-02 by Samahu