ouster-lidar/ouster-ros #60: Destaggerd Point Cloud

66lukas66, 2023-02-16:
I want the Ouster to have a destaggered (organized) Point Cloud published. As per the documentation located at https://static.ouster.dev/sdk-docs/reference/lidar-scan.html#staggering-and-destaggering, there is a function available for destaggering. However, I am unsure of where to implement it within the Ouster in order to obtain a destaggered point cloud.

My assumption is that the destaggering function needs to be called from within the scan_to_cloud_f function of the os_ros. But I am uncertain of the exact location within that function where the destaggering function should be integrated.
My guess is that it needs to be called inside the scan_to_cloud_f function of os_ros. But where exactly ?

Samahu, 2023-02-16:
So currently the destaggering step is only applied to the produced images, the code block that performs that can be found here:
https://github.com/ouster-lidar/ouster-ros/blob/22acf88f2f162cff62e10318aed7c51497ebc08a/src/os_image_nodelet.cpp#L137-L154
This code block is not using the destagger method since it would be much faster to do it this way when you are destaggering multiple images at the same time. rather than invoking the destagger method on each image.

So in essence, you can take the output of this method and convert it again to a PCL point cloud object and publish it on a separate topic.

On the other hand you can apply the same action or call it the destagger method before producing the first point cloud (i.e. modifying the results of the `/points` topic). If you choose this route then you should apply the destagger step on the range image before passing it to the `ouster::cartesianT` method on this line:
https://github.com/ouster-lidar/ouster-ros/blob/22acf88f2f162cff62e10318aed7c51497ebc08a/src/os_ros.cpp#L230

So you destagger the `range` image and pass the destaggered result to the `ouster::cartesian` method. It is worth noting that this will, however, affect the results of the `os_image_nodelet` since it will apply destaggering again and thus the image(s) will appear skewed if you try to visualize them in RVIZ.

Hope this answers your question

66lukas66, 2023-02-17:
Thanks for the answer but if in ouster-ros/src/os_ros.cpp before the range destagger, what parameter should pixel_shift_by_row be set to in the destagger method?

Samahu, 2023-02-17:
That parameter should always be set to `info.format.pixel_shift_by_row`, where **info** is the object you get from parsing the sensor metadata.

closed 2023-02-17 by Samahu