ouster-lidar/ouster-ros #61: Referece to #Destaggerd Point Cloud using of the destaggered function for Point Clouds

66lukas66, 2023-02-20:
I want the Ouster to have a destaggered (organized) Point Cloud published. As per the documentation located at https://static.ouster.dev/sdk-docs/reference/lidar-scan.html#staggering-and-destaggering, there is a function available for destaggering. 
I know that the destagger function must be used in os_ros.cpp before line 230.

But it still doesn't work. I destaggered the range before **line 230** in **ouster-ros/src/os_ros.cpp.**
```
Eigen::Ref<const ouster::img_t<uint32_t>> refrange = range;
range = ouster::destagger(refrange,pixel_shift_by_row);
```

But the published point cloud is still staggered. Is it because a staggered range is a timestep X scanline matrix and a destaggered range is an azimuth angle X scanline matrix. However, in the function **destagger(const Eigen::Ref<const img_t<T>>& img, const std::vector<int>& pixel_shift_by_row, bool inverse)** the size of the return matrix is ​​set to the size of **img**. 
As I understand it the return matrix is a matrix with the size timestep X scanline and not a destaggerd matrix with the size of azimuth angle X scanline matrix .
 Or is the Ouster azimuth angle number equal to the number of timesteps per scan? 
And what is the reason that only the range must be destaggerd and not also the reflectivity?

Samahu, 2023-02-22:
@66lukas66 I think the issue is that the `cartesian` method(s) which project the lidarscan to 3D expect the input 2D image to be in staggered state so you can not pass a destaggered image to and then expect the right results.

So what you should is first perform the cartesian then take the generated `points` and pass it to the destagger function. However, the destagger method expects `ouster::img_t` which is a 2D array with the same dimensions as the lidarscan {WxH}. Meanwhile, `points` is an array of the size {WxH, 3} so we beed to reshape it back to the WxH using the `Eigen::Map` function as follows:

```cpp
    auto points_reshaped = Eigen::Map<const ouster::img_t<float>>(points.col(0).data(), h, w);
```
then you can apply the destaggering method to the reshaped points as follows:
```cpp
    auto points_destaggered = ouster::destagger<float>(points_reshaped, info.format.pixel_shift_by_row);
```

Samahu, 2023-02-22:
> And what is the reason that only the range must be destaggerd and not also the reflectivity?

All the 2D images are being destaggered in `os_image_nodelet.cpp`, the destaggering is applied to 4 images at the same time within this code block:
https://github.com/ouster-lidar/ouster-ros/blob/927f2168f9e81b93c8904ab98c32ea3d54a73f12/src/os_image_nodelet.cpp#L139-L154

Hope my answers help

66lukas66, 2023-02-23:
Both of your ideas work, not perfectly, according to my implimentation. They don't always sort all items in the correct order. I have now written a function for detagging the cloud, which also transcribes the point cloud at the same time, which is based on the destagger function.

I'll make a pull request for it shortly.

Samahu, 2023-02-23:
Ok, I will take a look at your PR.

PR #188 (2023-08-10): SW-5396: publish point cloud in destaggered form
## Related Issues & PRs
- ported from #187
- related #60 
- closes #61 
- related #66 

## Summary of Changes
- Publish de-stagggered point cloud
- wrap all classes within the **ouster_ros** namespace

## Validation
- Review test procedures
PR #187 (2023-08-10): SW-5396: publish point cloud in destaggered form (foxy)
## Related Issues & PRs
- ported from #182
- related #60 
- closes #61 
- related #66 

## Summary of Changes
- Publish de-stagggered point cloud
- wrap all classes within the **ouster_ros** namespace
- add a `viz-reliable.rviz` configuration and use by default for record/replay senarios.

## Validation
- Review test procedures
closed 2023-08-10 by Samahu