ouster-lidar/ouster-ros #62: Merger for clouds of several lidars

ShepelIlya, 2023-02-21:
As i mentioned in [this comment](https://github.com/ouster-lidar/ouster_example/pull/278#issuecomment-1407430958) we use some other features for our work with ouster ros driver.
The main feature is `os_cloud_merger_nodelet`. This nodelet can merge clouds from several os_cloud_nodelets with standart ROS ApproximatePolicy Synchronizer. Also we can transform this cloud to preferred frame and use box filter on it. We use it on our robot with 2 ouster sensors on it, may be other users with multi-lidars setup need something like this

I can make a PR if you interested in this functionality.

Samahu, 2023-02-21:
That would be useful but I wouldn't be able to support with merge currently since I am occupied with ROS2 driver stability and optimization work, so it won't be until next quarter that I can take a look at this. I will keep this ticket open and we can circle back on it.

stephen-derosa, 2023-08-14:
Curious if there has been any update on this - has time opened up for this ?

Samahu, 2023-08-15:
Nope, not yet.