ouster-lidar/ouster-ros #65: How to work with the transforms in ros2 driver for rolling?

hodnajit, 2023-02-23:
Hi!
Thank you for your effort with the ros2 rolling driver for the ouster. I am happy to test it, and I have some doubts and ideas to share. :)

I setup the config file with this setting:

```
  ros__parameters:
    sensor_hostname: '10.64.1.6'
    udp_dest: '10.64.1.5'
    lidar_mode: '1024x10'
    timestamp_mode: 'TIME_FROM_ROS_TIME'
    udp_profile_lidar: ''
    metadata: ''
    lidar_port: 7502
    imu_port: 7503
ouster/os_cloud:
  ros__parameters:
    tf_prefix: ''
    timestamp_mode: '
```

And launched

`ros2 launch ouster_ros sensor.composite.launch.py`

Everything looks fine :)

I tried to integrate it to my robot, and I set in my urdf file the link `os_sensor` according to this (the whole urdf is in the attachment)
[model.txt](https://github.com/ouster-lidar/ouster-ros/files/10813685/model.txt)

```
<link name="os_sensor">
    <inertial>
      <mass value="1.1797214877679865"/>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <inertia ixx="0.0010891778635817933" ixy="0.0" ixz="0.0" iyy="0.0010891778635817933" iyz="0.0" izz="0.0011161639926144861"/>
    </inertial>
    <visual>
      <geometry>
        <mesh filename="file:///home/hodnajit/rtabmap_experiments/install/optimus_description/share/optimus_description/meshes/sensors/ouster_os0.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <material name="yellow"/>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="file:///home/hodnajit/rtabmap_experiments/install/optimus_description/share/optimus_description/meshes/sensors/ouster_os0.stl" scale="0.001 0.001 0.001"/>
      </geometry>
    </collision>
  </link>
  <joint name="ouster_joint" type="fixed">
    <origin rpy="0 0.06283 0" xyz="0 0 0.9175"/>
    <child link="os_sensor"/>
    <parent link="base_link"/>
  </joint>
```

I launch it
`ros2 launch ouster_ros sensor.composite.launch.py viz:=False`

I have no problem with tf if I visualisate it in frame base_footprint
[image]
(see the fixed_frame set to base_footprint and Pointcloud2 to topic /ouster/points)

But if I change the view to /odom or /map frame, it says there is no transform between /odom (or /map) to /os_imu or /os_lidar and I cannot see the pointcloud

[image]

What am I doing wrong? I tried to add the transforms os_imu and os_lidar into urdf file as well, but the result was the same :/. And I can see, there is a tf publisher in os_cloud

```
tf_bcast.sendTransform(ouster_ros::transform_to_tf_msg(
            info.imu_to_sensor_transform, sensor_frame, imu_frame, msg_ts));
```
and 
```
tf_bcast.sendTransform(ouster_ros::transform_to_tf_msg(
            info.lidar_to_sensor_transform, sensor_frame, lidar_frame, msg_ts));
```
So it looks to me that I do not need to publish the transform between sensor_frame and lidar(or imu)_frame.

Can you add the setting of sensor_frame as parameter? Or maybe the imu_frame and the lidar_frame as well? Or how does it work?

Thank you very much,

Best regards

Jitka Hodna

hodnajit, 2023-02-23:
I am a bit confused why all <>_image topics don't have frames set :) What is the idea behind it? Would it be possible to add the functionality of not publishing these images? :)

Samahu, 2023-02-23:
I tried to imitate your case by adding a static tf broadcaster between `map` and `os_sensor` but I couldn't get the same error that you are getting, here is a snapshot:
[image]

What does your TF tree look like in RVIZ if you expand it? (like I did in the above snapshot)

Both your urdf file and the rendered `frames.pdf` look poper to me. I would probably need to dig a little more into it to find out. 

> I am a bit confused why all <>_image topics don't have frames set :) What is the idea behind it? Would it be possible to add the functionality of not publishing these images? :)

Regarding this part of the question, yes you are right we should fill the `frame_id` when publishing these images (I will try to address this soon). On the other hand, if you don't care for these images you can completely drop them of your view and save on CPU too by commenting out the following line from the launch file that you are using:
https://github.com/ouster-lidar/ouster-ros/blob/7da1a96577bae52cece1801fc58907b0f3bcf8df/ouster-ros/launch/sensor.composite.launch.py#L71

hodnajit, 2023-02-27:
Thanks, the problem is on my side. I have an issue with `Lookup would require extrapolation into the past` when searching for tf between map --> os_imu/os_lidar. I am thinking about this question as it is done. Only one thing, can you please add the possibility of changing the frames names by parameter: os_sensor, os_imu and os_lidar? :) It would definitely help with integrating your driver to any robotic system :)

Samahu, 2023-02-27:
I had different users asking to be able to change frame names. But what is the motivation if I may ask? We do have the `tf_prefix` parameter, does this not avoid name collision?

hodnajit, 2023-02-27:
> I had different users asking to be able to change frame names. But what is the motivation if I may ask? We do have the `tf_prefix` parameter, does this not avoid name collision?

Yes, that is great, that there is a possibility to have tf_prefix  and yes it avoids the naming collision. My motivation to change the frame is, that in my project we have plenty of robots with plenty of sensors and we have naming conventions for sensors frames both in real and simulated worlds. We would like to keep our naming convention in integrated driver :) tf_prefix is not enough for us.

PR #170 (2023-07-25): SW-5148: back port improvements in ros 2 to ros 1
## Related Issues & PRs
- original PR: #157 
- related: #146 
- related: #149
- related: #156 
- related: #162
- related: #163 
- resolves: #8 
- resolves: #33 
- resolves: #65 for custom frame names.
- resolves: #76
- resolves: #105 and #9 by implemented a switch between static and dynamic broadacasts of TF messages. 
- resolves: #110
- resolves: #137
- resolves: #144
- resolves:  #155
- incorporates: #127
- resolves: #144
- resolves: #147

## Summary of Changes
- Added the **ThreadSafeRingBuffer**
  - with minimal unit tests for it
- Switch t [... cut]
closed 2023-02-27 by hodnajit