ouster-lidar/ouster-ros #68: fatal error: tf2_eigen/tf2_eigen.hpp: No such file or directory

reason-rock, 2023-02-26:
On ROS2 foxy, when build I got an error

ros2_ws/src/ouster-ros/ouster-ros/src/os_ros.cpp
fatal error: tf2_eigen/tf2_eigen.hpp: No such file or directory
   18 | #include <tf2_eigen/tf2_eigen.hpp>

I checked the packages blow, but everything already installed.

sudo apt install -y             \
    ros-$ROS_DISTRO-pcl-ros     \
    ros-$ROS_DISTRO-tf2-eigen   \
    ros-$ROS_DISTRO-rviz2

Samahu, 2023-02-26:
If you are using foxy distro of ROS2 you should use the ros2-foxy branch of
this repo, you are definitely on the ros2 branch. It is my fault that I
forgot to customize the readme for this branch to specify checking out
ros2-foxy instead of ros2. I will fix that tomorrow but for now you need to
checkout the repo as follows:

git clone -b ros2-foxy --recurse-submodules
https://github.com/ouster-lidar/ouster-ros.git

Samahu, 2023-02-27:
I have updated the README file to checkout the appropriate branch for ros2 foxy. Hope that my answer solved the build issue on your end. I am closing this ticket but let me know if you are still having trouble building the ouster driver on ros2 foxy distro.

closed 2023-02-27 by Samahu

reason-rock, 2023-03-01:
Thank you for your prompt reply and correction.
I saw the ros-foxy branch at that time, but I thought it would be just a temporary branch for development.
Now it works good, thanks again.
Have a good day :D