ouster-lidar/ouster-ros #70: Is there a necessity to have separete msg and srv pkgs?

DrShIkIgAmy, 2023-02-28:
Hi, I wonder if it really good practice to have separate msg and srv pkgs in project. As I see, the messages and services are only used within ouster-driver. That makes me think, that msg and srv can be defined in the ouster-driver pkg. That way we can reduce count of pkgs from three to two (lets not taking into account sdk submodule). If current structure of projects was not made for purpose, I can help to combine msg and srv into ouster-ros pkg

ShepelIlya, 2023-02-28:
Yeah, @Samahu i think that code structure that we have in ROS1 is more understandable. 
So can we switch 
- ouster-ros
- ouster-msgs
- ouster-srvs

To 
- ouster-ros
- - ouster-msgs
- - ouster-srvs
?

hodnajit, 2023-02-28:
According to [REP 144](https://www.ros.org/reps/rep-0144.html#id13) :)

> if a package contains any of a ROS message/service/action, append msgs

I think it is great, if these 2 packages ouster-ros and ouster-msgs are separated. But I think you can merge ouster-msgs and ouster-srvs together.

ShepelIlya, 2023-02-28:
> According to [REP 144](https://www.ros.org/reps/rep-0144.html#id13) :)
> 
> > if a package contains any of a ROS message/service/action, append msgs
> 
> I think it is great, if these 2 packages ouster-ros and ouster-msgs are separated. But I think you can merge ouster-msgs and ouster-srvs together.

I mean in ROS1 version we have _only one_ package with msgs and srvs in it. Here we have 3 different packages in one rep - so our suggestion is to combine this into 1 package as it was.

Samahu, 2023-02-28:
I think it is necessary to have the messages be in a separate package for the auto generated code to work properly. Services on the other hand don't need to be, I can see `ouster-srvs` merged into `ouster-msgs` but not into `ouster-ros`. I need to research this a bit and make sure things can cross compile if we merge `ouster-srvs` and `ouster-msgs`. But may I ask wwhat is the exact pain point of having multiple packages?

DrShIkIgAmy, 2023-03-01:
> I think it is necessary to have the messages be in a separate package for the auto generated code to work properly. Services on the other hand don't need to be, I can see `ouster-srvs` merged into [...]

Indeed, there's no some extra pain in it) But in my opinion, this is an incorrect approach to have a separate pkg for the messages, which are used locally only. The point is, that ouster_driver as a systems takes no input and provides output with standart msgs, so we don't use custom msgs. And It looks weird to me, why we have separated msg and srv, while it can be easily merged into driver pkg.

dambrosio, 2023-05-20:
### My two cents on this topic

I have found it extremely helpful to separate the driver interfaces (msg/srv) and driver implementation into different packages. The main reason is to reduce the dependencies on building the driver when a package only needs the interfaces. I know there has been a discussion about the current messages and services only being used locally, but I don't think we can assume that will always be the case for **all** applications. Imagine the case where a node in an external package is in charge of monitoring the ouster driver and needs access to the `SetConfig.srv` or `GetConfig.srv`, we wouldn't want the external package to build the entire `ouster_ros` package.

As per naming the msg/srv package: As I understand it REP 144 is outdated for ROS2, so maybe follow the style proposed in the official ROS2 tutorial on "[implementing custom interfaces](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Single-Package-Define-And-Use-Interface.html)"? This would look something like a package name `ouster_interfaces` with both msg and srv folders. This would correspond to imports and namespaces that explicitly list the interface type (ie. `ouster_interfaces::msg::PacketMsg`).
Since the ROS1 message already has a different package name (`outrider_ros`) compared with ROS2 (`outrider_msgs`) any sort of ROS1-ROS2 bridge would need a set of custom [mapping rules](https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst) anyways.

Samahu, 2023-05-20:
> ### My two cents on this topic
> I have found it extremely helpful to separate the driver interfaces (msg/srv) and driver implementation into different packages. The main reason is to reduce the [...]

@dambrosio Thanks for the input, I would like to re-visit the issue once I have the time.

Samahu, 2023-09-04:
Closing the issue as it was addressed by #176 and #177.

PR #176 (2023-07-28): SW-5345: merge-ouster-srvs into ouster-msgs
## Related Issues & PRs
- resolves: #70 

## Summary of Changes
- Merge ouster-srvs into ouster-msgs
- Update Dockefile
- Other fixes and code improvements
  - Create the lidar byproduct topics only when they are activated through proc_mask  

## Validation
The project builds successfully as usual
PR #177 (2023-07-28): SW-5345: merge ouster srvs into ouster msgs foxy
## Related Issues & PRs
- resolves: #70 

## Summary of Changes
- Merge ouster-srvs into ouster-msgs
- Update Dockefile
- Other fixes and code improvements
  - Create the lidar byproduct topics only when they are activated through proc_mask  

## Validation
The project builds successfully as usual
closed 2023-09-04 by Samahu