ouster-lidar/ouster-ros #80: Latched topic transport for config

TobinHall, 2023-03-11:
Hi, 
For a long time we have been using a fork of ouster_example as our ros driver, so it's nice to see this repo. 

One of our modifications to ouster_example was to add a latched std_msgs/String topic for transporting the config.
This message was published from os_node (now sensor_nodelet I believe) and simply contained the string returned from the get_metadata function.

The consumers were slightly less straight forward. The services were also available, so the service client and the subscriber essentially raced to set the config. Whichever managed to get the config first set the config - this was always the topic. Obviously this behaviour could be different.

The advantage of having latched topics is that they can be captured by rosbag. During replay, os_node (sensor_nodelet) is not required at all, and you do not need to manage the config file. I almost always forgot to capture this file with our bags. 

The cost of adding this seems only to be an additional topic, which seems worthwhile in my opinion. The services are still available for other nodes that consume packet messages.

I have not yet integrated our changes into this repo, but I can do so next week. I can create a pull request if desired, however the changes are fairly straight forward. 

Thanks.

Samahu, 2023-03-11:
Hi TobinHall,

I was actually considering to add a topic for some of the reasons you
stated and I have another reason specific to ROS2.

I think there was another person who submitted a PR for this a while back (
https://github.com/ouster-lidar/ouster_example/pull/288) but we did not
support it back then. Personally, I am in favor of this idea and will
support you (to get it merged) if you submit a PR.

Thanks
Ussama

TobinHall, 2023-03-11:
Hi Ussama, 

Ah, I did not check for duplicate issues on ouster example. 

It seems like the drawback for the implementation in  [ouster-lidar/ouster_example#288](https://github.com/ouster-lidar/ouster_example/pull/288) is that it would be less consistent with other implementations in ouster_example. With the repo split this may be less relevant. However, it is possible to a metadata topic transport as an alternative rather than a replacement.

The services and file saving can be retained, which would maintain compatibility.
The only issue is determining the desired behaviour on the client/subscriber side with regard to service or topic precedence.
I don't see this as being much of an issue, in (almost) every case the data will be the same regardless of the transport. 

Obviously the metadata file / services could be parameterised or deprecated in the future if desired.

Thanks, 
Tobin.

Samahu, 2023-03-13:
Yes, we want to retain the service for compatibility reasons. The way I see it would be that dependent nodes (like **os_image** and **os_cloud**) would watch for the metadata message and configure/reconfigure themselves based on that.

TobinHall, 2023-03-13:
> Yes, we want to retain the service for compatibility reasons. The way I see it would be that dependent nodes (like **os_image** and **os_cloud**) would watch for the metadata message and configure/reconfigure themselves based on that.

Sure. The implementation in #81 currently has a subscriber that lasts until a config has been received. This is the same behaviour as the using the service. It can be changed to update the config when a new message is received. 
Now that there is a mechanism to update or overwrite the config, it makes more sense now to have a long lived subscriber.

Samahu, 2023-04-12:
Marking this issue as completed, @TobinHall thanks for your contribution!

closed 2023-04-12 by Samahu