ouster-lidar/ouster-ros #82: In "/ouster/imu" topic, oritentation values are 0 all the time.

erenduman2, 2023-03-13:
Hello,
I am trying to find orientation values from  /ouster/imu when i listen the topic with "rostopic echo", orientation values always stay 0 even though I change the direction it looks. What can be the problem here?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
- Ouster Sensor? \[e.g. OS-0, OS-1, ..\]
- Ouster Firmware Version? \[e.g. 2.3, 2.4, ..\]
- ROS version/distro? \[e.g. noetic, foxy, humble, ..\]
- Operating System? \[e.g. Linux, Windows, MacOS\]
- Machine Architecture? \[e.g. x64, arm\]
- git commit hash (if not the latest).

Samahu, 2023-04-01:
The sensor only generates acceleration and angular velocity, and the driver is publishing these raw values as is. We would need to do further processing to produce orientation values.

I'll keep this issue open until we have time to address it.

Ab-Tx, 2025-05-13:
Is there any expected release-date for this functionality?
I'm using OS0 Rev7, and the [device dataset](https://data.ouster.io/downloads/R/datasheet-rev7p1-os0.pdf) states it is available, so I wonder if the sensor is already sending the information and the ROS package is not using it.

Samahu, 2025-05-20:
Hi @Ab-Tx yes the imu device itself is equipped with a magnetometer but it is turned off. The reason being it is sustainable to interference with the sensor so we have that disabled. You would well use the code on that I have on the branchhttps://github.com/ouster-lidar/ouster-ros/pull/401 which uses simpler filter to populate the orientation values.