ouster-lidar/ouster-ros #83: sometimes the lifecycle cannot be set activate automatically

ahuazuipiaoliang, 2023-03-21:
**Describe the bug**
When I launch the ros2 driver, I cannot get pointcloud msgs or IMU msgs, then I have to activate the lifecycle manually by typing `ros2 lifecycle set ouster/os_sensor activate`.
I've seen the driver executes the command above, but it has no effect.

`software version`: [fd76cc9](https://github.com/ouster-lidar/ouster-ros/commit/fd76cc9eb0588590094ee012e9312bda276518da)
`hardware`: OS-1-32-U2
`firmware`: 
[image]

**Platform (please complete the following information):**
- Ouster Sensor?  OS-1-32-U2
- Ouster Firmware Version?  v2.3.1
- ROS version/distro?  humble
- Operating System?  Ubuntu 22.04
- Machine Architecture?  x64
- git commit hash [fd76cc9](https://github.com/ouster-lidar/ouster-ros/commit/fd76cc9eb0588590094ee012e9312bda276518da)

Samahu, 2023-03-21:
Hi Mark, when this happens what does the following command produce?

ros2 lifecycle get ouster/os_sensor

Thanks
Ussama

ahuazuipiaoliang, 2023-03-21:
launch file
``` xml
<group>
      <push-ros-namespace namespace="top"/>
      <arg name="ouster_ns" default="ouster"
      description="Override the default namespace of all ouster nodes"/>
    <arg name="sensor_hostname" default="192.168.1.120"
      description="hostname or IP in dotted decimal form of the sensor"/>
    <arg name="udp_dest" default=""
      description="hostname or IP where the sensor will send data packets"/>
    <arg name="lidar_port" default="0"
      description="port to which the sensor should send lidar data"/>
    <arg name="imu_port" default="0"
      description="port to which the sensor should send imu data"/>
    <arg name="udp_profile_lidar" default=""
      description="lidar packet profile; possible values: {
      LEGACY,
      RNG19_RFL8_SIG16_NIR16_DUAL,
      RNG19_RFL8_SIG16_NIR16,
      RNG15_RFL8_NIR8
      }"/>
    <arg name="lidar_mode" default=""
      description="resolution and rate; possible values: {
      512x10,
      512x20,
      1024x10,
      1024x20,
      2048x10,
      4096x5
      }"/>
    <arg name="timestamp_mode" default="TIME_FROM_ROS_TIME"
      description="method used to timestamp measurements; possible values: {
      TIME_FROM_INTERNAL_OSC,
      TIME_FROM_SYNC_PULSE_IN,
      TIME_FROM_PTP_1588,
      TIME_FROM_ROS_TIME
      }"/>
    <arg name="metadata" default=""
      description="path to write metadata file when receiving sensor data"/>
    <arg name="tf_prefix" default="" description="namespace for tf transforms"/>
  
    <group>
      <push-ros-namespace namespace="$(var ouster_ns)"/>
      <node_container pkg="rclcpp_components" exec="component_container_mt" name="os_container" output="screen" namespace="">
        <composable_node pkg="ouster_ros" plugin="ouster_ros::OusterSensor" name="os_sensor">
          <param name="sensor_hostname" value="$(var sensor_hostname)"/>
          <param name="udp_dest" value="$(var udp_dest)"/>
          <param name="lidar_port" value="$(var lidar_port)"/>
          <param name="imu_port" value="$(var imu_port)"/>
          <param name="udp_profile_lidar" value="$(var udp_profile_lidar)"/>
          <param name="lidar_mode" value="$(var lidar_mode)"/>
          <param name="timestamp_mode" value="$(var timestamp_mode)"/>
          <param name="metadata" value="$(var metadata)"/>
        </composable_node>
        <composable_node pkg="ouster_ros" plugin="ouster_ros::OusterCloud" name="os_cloud">
          <param name="tf_prefix" value="$(var tf_prefix)"/>
          <param name="timestamp_mode" value="$(var timestamp_mode)"/>
        </composable_node>
        <composable_node pkg="ouster_ros" plugin="ouster_ros::OusterImage" name="os_image"/>
      </node_container>
    </group>
  
    <!-- HACK: configure and activate the sensor node via a process execute since state
      transition is currently not availabe through launch.xml format -->
    <executable cmd="$(find-exec ros2) lifecycle set sensing/lidar/top/$(var ouster_ns)/os_sensor configure"
      launch-prefix="bash -c 'sleep 0; $0 $@'" output="screen"/>
    <executable cmd="$(find-exec ros2) lifecycle set sensing/lidar/top/$(var ouster_ns)/os_sensor activate"
      launch-prefix="bash -c 'sleep 1; $0 $@'" output="screen"/>
    </group>
```

ahuazuipiaoliang, 2023-03-21:
> Hi Mark, when this happens what does the following command produce? ros2 lifecycle get ouster/os_sensor Thanks Ussama
> -- *Ussama Naal* San Francisco, CA <http://ouster.com/>

I got `inactive [2]`

ahuazuipiaoliang, 2023-03-21:
Sometimes I got `active [3]`.

Samahu, 2023-03-21:
You could try switching to the python based launch files and see if this works better for you:
```bash
ros2 launch ouster_ros sensor.independent.launch.py param_files:=<path-to-your-param-files>
```
or
```bash
ros2 launch ouster_ros sensor.composite.launch.py param_files:=<path-to-your-param-files>
```

ahuazuipiaoliang, 2023-03-21:
> You could try switching to the python based launch files and see if this works better for you:
> 
> ```shell
> ros2 launch ouster_ros sensor.independent.launch.py param_files:=<path-to-your-param-files>
> ```
> 
> or
> 
> ```shell
> ros2 launch ouster_ros sensor.composite.launch.py param_files:=<path-to-your-param-files>
> ```

Yes, I will try.
BTW, can we activate the lifecycle without setting manually?

Samahu, 2023-03-21:
> BTW, can we activate the lifecycle without setting manually?

What do you mean? I am not sure that I quite understood your question? You would have to cycle through the node states till you reach the active(activated) state. That's what all launch files are doing automatically for you. The issue you are facing is that launch.xml files don't have a mechanism to do transition when it is approprite, the py launch files do have this ability, but they are a departure from what our users currently depend on and I haven't tested them thoroughly but they should work. Let me know if these launch files work better for you than the launch.xml files.

Samahu, 2023-04-07:
> Thanks, I will try.

Following up on this issue, did you get to out the two launch files?

> BTW, can we activate the lifecycle without setting manually?

All the launch files do activate the lifecycle automatically, however, the python based launch are event based while xml ones don't have this feature and the state change is requested after a delay

Samahu, 2023-05-15:
@Andre-ex9 thanks for reporting your experience we may have a fix soon, so stay tuned.

Samahu, 2023-07-20:
This has been addressed by the #146. If you use the default `sensor.launch.xml` or `driver.launch.py` this would utilize the combined node\component `os_driver` rather than the three components (`os_sensor`, `os_cloud` and `os_image`). This simplifies the initialization of the driver and has other benefits.

`sensor.launch.xml` however, still relies on the command line invocation to update the node lifecycle. So if this still causing you an issue then you would be better off with `driver.launch.py` which is event based.

Please try it out and let me know if we can close this issue.

Samahu, 2023-07-24:
@ahuazuipiaoliang or @Andre-ex9 any chance that you have tried the updated driver? If I don't hear back on the issue within this week I am going to mark this this ticket as done. Please let me know of the update described in my previous comment has helped mitigate the problem.

Samahu, 2023-07-25:
Closing the issue due to inactivity, feel free to re-open if the issue is still present.

closed 2023-07-25 by Samahu