ouster-lidar/ouster-ros #84: Build error

hichamhendy, 2023-03-24:
**Describe your question**
A clear and concise description of your question.
I am getting the error by `/usr/include/eigen3/Eigen/src/Core/util/ConfigureVectorization.h:40:38: note: attribute for ‘struct pcl::_PointXYZHSV’ must follow the ‘struct’ keywor`
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Platform (please complete the following information):**
- Ouster Sensor? OS-1
- Ouster Firmware Version? The latest
- ROS version/distro? Melodic
- Operating System? Linux
- Machine Architecture? x64,

Samahu, 2023-03-24:
What is the version of Eigen library installed on your system? (Try `dpkg -l | grep eigen`)

Samahu, 2023-03-25:
Another thing that could be a factor is the compiler version, can you report these two values so we can further help.

hichamhendy, 2023-03-25:
Regarding the compiler,  am using gcc 11. 
regarding the eigen using 3.3

Samahu, 2023-03-27:
What is the subversion of Eigen? I built the project successfully on Ubuntu18/melodic with Eigen 3.3.4 installed.

Samahu, 2023-03-29:
@hichamhendy I haven't heard back from you for couple of days. You didn't mention the minor version of the eigen libdary that is installed on your system. If your problem still persist I recommend that you try to upgrade your eigen library installation to the latest or maybe uninstall and re-install. On Ubuntu 18.04 you should have at minimum the version 3.3.4 of the eigen library on your system and then try to re-compile the project.

I am going to close this issue for now but if you feel that I haven't addressed your question feel free to re-open.

Best
Ussama

closed 2023-03-29 by Samahu

Samahu, 2023-04-25:
Eigen 3.4.0 should be fine. I would look some where else for the problem. Don't know how to help since I am unable to re-produce the issue and this problem doesn't seem to be very common among our user base. What you could try:
- Try running the driver within docker
- Remove ROS and Eigen and then re-install.