ouster-lidar/ouster-ros #85: Service calls for get_metadata and get_config not working.

Grisly00, 2023-03-24:
Launching the driver works fine and I get data in rviz2 but when I try to invoke a service call as described in the readme I get the error: 
```
The passed service type is invalid
```
  
If I check the type I get:
```
$ros2 service type /ouster/get_metadata
ouster_srvs/srv/GetMetadata
```

**To Reproduce**
```
$ros2 launch ouster_ros sensor.launch.xml sensor_hostname:=192.168.1.60 lidar_port:=7502 imu_port:=7503 udp_dest:=192.168.1.110 udp_profile_lidar:=RNG19_RFL8_SIG16_NIR16_DUAL
```
```
$ros2 service call /ouster/get_metadata ouster_srvs/srv/GetMetadata
```
or
```
$ros2 service call /ouster/get_config ouster_srvs/srv/GetConfig
```

**Platform (please complete the following information):**
- Ouster Sensor? OS1
- Ouster Firmware Version? 2.3.0
- ROS version/distro? Humble
- Operating System? Ubuntu Jammy (in WSL)
- Machine Architecture? x86_64
- Branch: ros2

Samahu, 2023-03-24:
Do you source the `install/setup.bash` in the terminal where you execute the service call?

Grisly00, 2023-03-25:
I am very sorry for that..

closed 2023-03-25 by Grisly00