ouster-lidar/ouster-ros #86: Build problem

hichamhendy, 2023-03-24:
**Describe the bug**
by building the package I get `/opt/ros/melodic/include/tf2_eigen/tf2_eigen.h:188:66: error: implicitly-declared ‘tf2::Stamped<Eigen::Matrix<double, 3, 1> >& tf2::Stamped<Eigen::Matrix<double, 3, 1> >::operator=(const tf2::Stamped<Eigen::Matrix<double, 3, 1> >&)’ is deprecated [-Werror=deprecated-copy]
  188 |                                         transform.header.frame_id);
`

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. source ros environment
2. compile the project workspace
3. source the project workspace
4. ros launch ouster_ros sensor or replay
5. open another terminal and the project workspace
6. observe the issue ...

**Screenshots**
[image]

**Platform (please complete the following information):**
- Ouster Sensor? OS-1
- Ouster Firmware Version? Latest
- ROS version/distro? melodic
- Operating System? Linux
- Machine Architecture? x64

Samahu, 2023-03-29:
@hichamhendy This is likely an issue related to the other issue #84 that you have created, the issue seems specific to your current setup of ros/eigen libraries. I think the compiler version is less of an issue here and I did verify that the project compiles successfully on an ubuntu:18.04 machine with the following combination of the specific libraries:
```
libeigen3-dev                          3.3.4
ros-melodic-tf2-eigen                  0.6.5
```

So maybe check what versions you have installed on your machine and try to match the same versions or upgrade to the latest. If the problem still persist you may need to re-try installing ROS and/or start with a fresh system.

I am going to close the issue but if you feel that I have fully addressed your question feel free to re-open.

Best
Ussama

closed 2023-03-29 by Samahu

hichamhendy, 2023-04-24:
I have  the following 
`libeigen3-dev                                                    3.4.0-1~bionic `
`ros-melodic-tf2-eigen                                            0.6.5-0bionic.20221025.181036 `