ouster-lidar/ouster-ros #87: Rosbag play question OS0 sensor and how to set up orientation

Rotoslider, 2023-03-28:
I have a OSO-32 sensor that I am using in a slam application. I currently use "roslaunch ouster_ros record.launch" to record bag files and then "roslaunch ouster_ros replay.launch" to replay the bag on a different computer. While the bag is playing I run the slam software. Sometimes it would be handy to have the bag start some seconds into it or at a faster rate of playback. Both of these things can be done with rosbag play but I have not figured out a way to do it with replay. Any suggestions? 
Second question:
The sensor is mounted with the Y axis up. How do I rotate the sensor information before the bag file so the slam software reads Y as Z up. 
[image]

- Ouster Sensor OS-0
- Ouster Firmware Version2.2.1
- ROS version/distro noetic
- Operating System Linu
- Machine Architecture x64 and arm

Samahu, 2023-03-29:
> Sometimes it would be handy to have the bag start some seconds into it or at a faster rate of playback. Both of these things can be done with rosbag play but I have not figured out a way to do it with replay. Any suggestions?

The `replay.launch` doesn't give you the option to control how do you play the bag file. So what you can do is edit the line:
https://github.com/ouster-lidar/ouster-ros/blob/12a479ccceac753cd86adae6509b516fc157766f/launch/replay.launch#L40 

and manually add the parameters for start and playback speed:
```xml
  <node if="$(arg _use_bag_file_name)" pkg="rosbag" type="play" name="rosbag_play_recording"
    launch-prefix="bash -c 'sleep 3; $0 $@' "
    output="screen" required="true"
    args="--clock $(arg bag_file) --start 10 --rate 0.5"/> <!-- use any other values -->
```

>Second question: Can I put the lidar back in standby mode with a command at the end of the rosbag record or?

You can achieve this through the **set_config** service. Basically create a sensor configuration file (let's say some-config.json) with the following setting:
```json
{ "operating_mode": "STANDBY" }
```
Next invoke the set_config service as described in the [README](https://github.com/ouster-lidar/ouster-ros#setconfig-experimental). This should stop the sensor from streaming and put it in STANDBY mode

> Third question: The sensor is mounted with the Y axis up. How do I rotate the sensor information before the bag file so the slam software reads Y as Z up. [image]

You would need to create a parent link and then add the sensor as a child and you need to configured the proper relative orientation.

Samahu, 2023-03-29:
Please find my other answer [here](https://github.com/ouster-lidar/ouster-ros/issues/37#issuecomment-1344540371) which should only serve as hint, if this wasn't enough you could try to check out some ros tutorials and and/or reach out to the ros community https://answers.ros.org.

Samahu, 2023-03-31:
I will mark this issue as closed, if you feel that I haven't fully addressed your questions or concerns feel free to re-open or open a new issue.

closed 2023-03-31 by Samahu