ouster-lidar/ouster-ros #90: Unable to receive data from OS1-64

xYulx, 2023-03-29:
Hi, i am pretty new to ROS & LIDAR and is trying to record data into a bag data.
I have currently build and installed the ouster ros in my docker image.
There doesn't seems to be any error arising from the run, but i do not see any data in rviz or data from any topic.
Does anyone have similar issue or any idea how i can go about to troubleshoot this?
i have tried to run the sensor.lauch and check the topic list, i tired to subscribe to the topic, /ouster/points but there doesnt seem to be any data.

**Ouster Sensor:** OS1-64
**Ouster Firmware Version:** ousteros-image-prod-aries-v2.1.3+20211021192601.patch-v2.1.3
**ROS version:** Melodic
**Operating System:** Linux
**GPU:** None

Below are some of the screenshot of the execution:
`roslaunch ouster_ros sensor.launch sensor_hostname:=192.168.100.51`

[image]
[image]

`rostopic list`

[image]

`rostopic hz /ouster/points /ouster/points2 /ouster/imu /ouster/range_image /ouster/reflec_image /ouster/signal_image`

[image]

Samahu, 2023-03-29:
@xYulx thanks for contacting us. To be able to work with the sensor from within the docker container you would need to launch the container with the network mode set to host. For example:
```bash
docker run [other-options] --network host your-docker-image
```

Hope this helps
Ussama

closed 2023-03-29 by Samahu

xYulx, 2023-03-30:
it works, thanks a lot for the help.
👍