ouster-lidar/ouster-ros #91: message_filter.ApproximateTimeSynchronizer method is not able to invoke the callback function for Ouster os1 lidar and usb camera

deepumdev, 2023-03-29:
I am using the Ouster OS1 LIDAR with ROS Melodic. The ROS package is publishing the LIDAR data and I am able to view it in RViz. I want to calibrate and fuse the LIDAR data with data from a USB camera, so I have implemented the cv_camera ROS package for the camera and created a new package called calib. However, I am facing an issue where the message_filter.ApproximateTimeSynchronizer method is not able to invoke the callback function. The topics I am subscribing to are cv_camera/image_raw, cv_camera/camera_info, and ouster/points. When I try to echo the ouster/points topic, the timestamp is not showing. How can I synchronize the time between the data streams ?

deepumdev, 2023-03-30:
> The timestamp should be included in the body of the **PointCloud2** messages. Can you show me the output you get when executing the command `rostopic echo --noarr /ouster/points`?

[image]

Kindly go through the attachment.  It has the time stamp. But my code doesn't able to invoke the callback function. Could you please help me to sort out it.

marufino, 2023-03-30:
The timestamps don't match between your lidar data and camera data.

That's because the lidar is counting seconds from when it was turned on, and the camera is in ROS time

In the ouster sensor.launch file, set timestamp_mode to TIME_FROM_ROS_TIME and that should allow your timestamps to be close enough to be captured by the synchronizer

Samahu, 2023-03-30:
> The timestamps don't match between your lidar data and camera data.
>
> That's because the lidar is counting seconds from when it was turned on, and the camera is in ROS time
>
> In the ouster [...]

Good thought, this is likely the issue. @deepumdev When relying on **Time Synchronization** you need to handle make sure the included sensors are running in Sync (either using a dedicated PTP clock, or depend entirely on ROS time, and for that you would need `timestamp_mode:=TIME_FROM_ROS_TIME` to `sensor.launch` as @marufino suggested.

deepumdev, 2023-03-31:
> The timestamps don't match between your lidar data and camera data.
>
> That's because the lidar is counting seconds from when it was turned on, and the camera is in ROS time
>
> In the ouster [...]

Its working fine. Thank you.

closed 2023-03-31 by deepumdev