ouster-lidar/ouster-ros #92: Build error humble: failed to find exported target names

racingteamepfljetson, 2023-03-29:
**Describe the bug**
I am trying to build the ros2 humble node but i run into the following error:

```bash
Starting >>> ouster_msgs
Starting >>> ouster_srvs
Finished <<< ouster_msgs [2.84s]
Finished <<< ouster_srvs [3.07s]
Starting >>> ouster_ros
--- stderr: ouster_ros
CMake Error at /home/racingteam/ros2_ws/install/ouster_msgs/share/ouster_msgs/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/racingteam/ros2_ws/install/ouster_msgs/share/ouster_msgs/cmake/export_ouster_msgs__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /home/racingteam/ros2_ws/install/ouster_msgs/share/ouster_msgs/cmake/ouster_msgsConfig.cmake:41 (include)
  CMakeLists.txt:16 (find_package)

---
Failed   <<< ouster_ros [1.29s, exited with code 1]

Summary: 2 packages finished [4.94s]
  1 package failed: ouster_ros
  1 package had stderr output: ouster_ros
```
I have no idea what has gone wrong, all the mentioned files above exist.
Thank you in advance for your help.

**To Reproduce**
1. source ros installation, in my case activate the conda env (see below)

**Platform (please complete the following information):**
I am using an OS1 64 channels gen1, ubuntu 20.04 on a Jetson Orin AGX (arm64), and I installed ros2 humble in a conda environment via robostack (and all the dependencies, just as described in the readme). I checkout out the ros2 branch (fd76cc9eb0588590094ee012e9312bda276518da when I wrote this issue) for ouster-ros, and 202c536b358b537b3620964cea3919a24699a7dc for ouster-sdk.

Samahu, 2023-03-29:
@racingteamepfljetson Thanks for contact us, I suspect that you may have missed packages that weren't installed (if you used rosdep to install the dependency then retry to install requirements using the direct method), additionally could you do me a favor and try to clean up your workspace (by performing `rm -rf build install/ log/` within _ros2_ws_ folder) and then try to rebuild and see if that would fix the problem

racingteamepfljetson, 2023-03-30:
I tried several times deleting the build, log and install dirs but nothing changes. I have not used rosdep but here are the outputs of `ros2 pkg list` and `mamba list`:

<details> 

<summary> `mamba list` </summary>

```bash
# packages in environment at /home/racingteam/miniforge3/envs/brains:
#
# Name                    Version                   Build  Channel
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.8                h4e544f5_0    conda-forge
ampl-mp                   3.1.0             h05efe27_1006    conda-forge
aom                       3.5.0                headf329_0    conda-forge
argcomplete               3.0.5              pyhd8ed1ab_0    conda-forge
assimp                    5.2.4                hf7258a5_0    conda-forge
atk-1.0                   2.38.0               hf4e84e4_1    conda-forge
attr                      2.5.1                h4e544f5_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
binutils                  2.39                 h64c2a2e_1    conda-forge
binutils_impl_linux-aarch64 2.39                 h48546ad_1    conda-forge
binutils_linux-aarch64    2.39                h489c705_12    conda-forge
black                     23.1.0           py39h4420490_0    conda-forge
boost                     1.74.0           py39ha253240_5    conda-forge
boost-cpp                 1.74.0               ha1c1135_8    conda-forge
brotli                    1.0.9                h4e544f5_8    conda-forge
brotli-bin                1.0.9                h4e544f5_8    conda-forge
bullet                    3.21                 hdae49d8_4    conda-forge
bullet-cpp                3.21             py39h2920bb4_4    conda-forge
bzip2                     1.0.8                hf897c2e_4    conda-forge
c-ares                    1.18.1               hf897c2e_0    conda-forge
c-compiler                1.5.2                hb4cce97_0    conda-forge
ca-certificates           2022.12.7            h4fd8a4c_0    conda-forge
cairo                     1.16.0            hd19fb6e_1014    conda-forge
casadi                    3.5.5           py39h60ee229_18    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39hb26bf21_3    conda-forge
cfgv                      3.3.1              pyhd8ed1ab_0    conda-forge
click                     8.1.3           unix_pyhd8ed1ab_2    conda-forge
cmake                     3.24.3               hc6ca916_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39h4420490_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compilers                 1.5.2                h8af1aa0_0    conda-forge
console_bridge            1.0.2                hdd96247_1    conda-forge
contourpy                 1.0.7            py39hd9a2fea_0    conda-forge
coverage                  7.2.2            py39h599bc27_0    conda-forge
cppcheck                  2.10.3           py39h6b9cb42_0    conda-forge
cryptography              39.0.0           py39h5e01b12_0    conda-forge
[... 930 lines cut ...]
tf2_ros
tf2_ros_py
tf2_sensor_msgs
tf2_tools
tinyxml2_vendor
tinyxml_vendor
tlsf
tlsf_cpp
topic_monitor
tracetools
trajectory_msgs
turtlesim
uncrustify_vendor
unique_identifier_msgs
urdf
urdf_parser_plugin
visualization_msgs
yaml_cpp_vendor
zstd_vendor
```

</details>

You can see that `ros-humble-pcl-ros`, `ros-humble-rviz2` and `ros-humble-tf2-eigen` are all installed.
Should I try with rosdep?