ouster-lidar/ouster-ros #97: [ROS2] configurable pointcloud type

doisyg, 2023-04-03:
**Is your feature request related to a problem? Please describe.**
Generated pointcloud is too heavy with 9 fields for most use case:
- x
- y
- z
- intensity
- t
- reflectivity
- ring
- ambient
- range

> POINT_CLOUD_REGISTER_POINT_STRUCT(ouster_ros::Point,
>     (float, x, x)
>     (float, y, y)
>     (float, z, z)
>     (float, intensity, intensity)
>     // use std::uint32_t to avoid conflicting with pcl::uint32_t
>     (std::uint32_t, t, t)
>     (std::uint16_t, reflectivity, reflectivity)
>     (std::uint16_t, ring, ring)
>     (std::uint16_t, ambient, ambient)
>     (std::uint32_t, range, range)
> )

**Describe the solution you'd like**
Configurable pointcloud fields, or at least the option the choose the light PointXYZI format

dambrosio, 2023-05-20:
@doisyg that is a great idea. I also think one thing that could dramatically reduce the size of the output `PointCloud2` message would be to remove the unused/padding bytes in the `PointCloud2` `data` array.

For example, in a single `point_step` there are 18 bytes of padding:
- 4 bytes between the `z` and `intensity` fields
- 2 bytes between the `ambient` and `range` fields
- 12 bytes after the `range` field
-
This would change each `point_step` to 30 instead of 48.

For simple math, a PointCloud with 100,000 points would result in a `data` array size difference of ~46.15%. This would be 4.57MB vs 2.86MB per PointCloud2 message. 

Maybe there is a reason for padding each `point_step` but it does increase the size of each PointCloud message a thus increases the cost of processing and transmitting the messages.

dambrosio, 2023-05-22:
Also, this looks to be a duplicate of #29, may want to combine.

PR #253 (2023-11-16): SW-5657: Back port SW-5466 to ROS-1
## Related Issues & PRs
- parent #216 
- resolves #15 
- resolves #29
- resolves #210 
- resolves #97
- related #217
- related #223
- resolves #231 (adds a note to CHANGLOG.rst indicating that the default `ptp_utc_tai_offset` forms a breaking change).  

## Summary of Changes
- added the ability to customize the published point clouds(s) to velodyne point cloud format and
  other common pcl point types.
- made ouster_image_nodelet operate separately from ouster_cloud_nodelet
- added a note to `CHANGLOG.rst` indicating that the default `ptp_utc_tai_offset` fo [... cut]
closed 2023-11-16 by Samahu