ouster-lidar/ouster-ros #105: High TF Broadcast Rate

outrider-jkoch, 2023-04-18:
I noticed in this commit (https://github.com/ouster-lidar/ouster-ros/commit/600c55cae143f98f62e283fa54eac8964177e695) that the transform was changed to support working with ROS time. Maybe I am misunderstanding, but I'm confused why a static transform (from metadata) needs to send an TF2 message for every received IMU and Lidar packet. I notice a large quantity of TF2 messages are being sent but never with dynamically changing data. Could this be throttled in anyway or is it necessary to send the TF2 message at this rate?

We have observed this with all of our sensors:
OS-0 and OS-1
Commit: https://github.com/ouster-lidar/ouster-ros/commit/3c2b6a1bb0eca0b400106cce4ea978deff18a46d

Samahu, 2023-04-18:
> I noticed in this commit (600c55c
> Commit: 3c2b6a1

Data isn’t update but timestamp is. Subscribers like RVIZ would complain
about the TF data being too old if we simply use static broadcaster. This
is the reason we send the TF messages. Currently, we are sending the TF
messages at the same frequency as the IMU and PointCloud messages (which
are 100Hz ane 10Hz respectively) but this is perhaps doesn’t need to be the
case. I would need to check if reducing the rate of TF messages to 1Hz
would be sufficient and wouldn’t cause a warning about data too old.

Aposhian, 2023-04-27:
rviz on ROS 2 doesn't warn when static transforms are being published. Even so, publishing messages at high rates is a significant computational expense, and shouldn't be done just to silence warnings. A static transform broadcaster is correct for unchanging transforms

Samahu, 2023-04-27:
The reason I update the TF messages even though the TF data don't change was due to [a concern raised by one of our users](https://github.com/ouster-lidar/ouster-ros/issues/9#issuecomment-1311528903). The message states:
```
[ WARN] [1668158414.957718441]: odometry: Could not get transform from base_link to os_sensor (stamp=1668158375.848681) after 0.500000 seconds ("wait_for_transform_duration"=0.500000)! Error="Lookup would require extrapolation 38.615832763s into the past. Requested time 1668158375.848681211 but the earliest data is at time 1668158414.464514017, when looking up transform from frame [os_sensor] to frame [base_link]. canTransform returned after 0.504406 timeout was 0.5."
```
It is a warning level message and does give the impression that there is a problem. It is one of those cases where ROS behavior is inconsistent.  I am not sure if the problem still exist in ROS2, I haven't tried. But if it doesn't we can drop the dynamic rebroadcast of TF messages.

Samahu, 2023-04-27:
Otherwise I am more inclined to keep current code as is but reduce the frequency of published TF messages to 2-10 Hz.

Aposhian, 2023-04-28:
Sending an unchanging value on TF instead of TF static, at least on ROS 2, is incorrect. If that is required to eliminate warnings on ROS, then it should be on ROS, but not ROS 2. ROS 2 should definitely use tf static for this. This is how it is done in the unofficial driver, and this plays nice with rviz and other tools on ROS 2. https://github.com/ros-drivers/ros2_ouster_drivers/blob/5425ed023d3c1c5293a9af90d2ae81c5df889334/ros2_ouster/src/ouster_driver.cpp#L138

Samahu, 2023-05-05:
@outrider-jkoch we have addressed the issue for the ROS2 driver, do you mainly use ROS1?

outrider-jkoch, 2023-05-10:
We are using ROS1 at the moment but we have some plans to use ROS2 in the future. Until then though having the rate for the TF broadcast reduced would be desirable.

Samahu, 2023-05-10:
> We are using ROS1 at the moment but we have some plans to use ROS2 in the future. Until then though having the rate for the TF broadcast reduced would be desirable.

Thanks for the feedback, please take a look at #134 and let me know if this seems like a reasonable compromise to you.

Samahu, 2023-05-12:
@outrider-jkoch I haven't got any feedback from you about the proposed solution #134 . Since I don't see any harm in it and that it does bring down the TF publish rate from 120+ Hz to 20-40Hz I went ahead and merged the solution. Let us know if the updated TF publish rate still poses a problem on your end.

closed 2023-05-12 by Samahu

outrider-jkoch, 2023-05-16:
I think this is an improvement from what it was. However, I'm still skeptical that switching from `static_transform_broadcaster` to `transform_broadcaster` was correct. As @Aposhian said with regards to ROS2, sending an unchanged transform on /tf versus /tf_static is incorrect. I didn't look much into the problem that was experienced in https://github.com/ouster-lidar/ouster-ros/issues/9 but there seems to be a problem with latching the transform in that context that would need more investigation.

Samahu, 2023-05-16:
> I think this is an improvement from what it was. However, I'm still skeptical that switching from `static_transform_broadcaster` to `transform_broadcaster` was correct. As @Aposhian said with [...]

Yeah, I am not arguing that but at the same time because of the issue reported in #9 - which I have experienced it myself when working with RVIZ - I am not inclined atm to change it back to static tf broadcaster for the ROS1 driver unless there is a compelling reason. We did the switch for the ROS2 version of the driver.

PR #124 (2023-05-05): SW-4972: merge switching to static transform publisher contribution
## Related Issues & PRs
- parent #112
- ros2 foxy #125
- resolves #105 for ros2

## Summary of Changes
- Send static transforms only once with a static transform broadcaster

## Validation
- [optional] Launch sensor node (xml | py) and replay / record nodes.
PR #125 (2023-05-05): SW-4972: merge switching to static transform publisher contribution foxy
## Related Issues & PRs
- parent #112
- ros2 rolling/humble #124
- resolves #105 for ros2

## Summary of Changes
- Send static transforms only once with a static transform broadcaster

## Validation
- [optional] Launch sensor node (xml | py) and replay / record nodes.
PR #134 (2023-05-12): SW-4993: Reduce the publish rate of imu tf transforms
## Related Issues & PRs
- closes #105 
- related #9 

## Summary of Changes
- Reduce the publish rate of tf transforms by publishing IMU TF transforms at the same rate as LIDAR TF transforms

> **Note:**
> Using **StaticTransformPublisher** is the proper thing to do but due to an issue that exists(or existed) in ROS1 which causes it to generate tons of warnings that couldn't be silenced as noted in #9 and also based on my own experience; so as a compromise I kept everything pretty much as is but reduced the publish rate of IMU TF transforms [... cut]
PR #170 (2023-07-25): SW-5148: back port improvements in ros 2 to ros 1
## Related Issues & PRs
- original PR: #157 
- related: #146 
- related: #149
- related: #156 
- related: #162
- related: #163 
- resolves: #8 
- resolves: #33 
- resolves: #65 for custom frame names.
- resolves: #76
- resolves: #105 and #9 by implemented a switch between static and dynamic broadacasts of TF messages. 
- resolves: #110
- resolves: #137
- resolves: #144
- resolves:  #155
- incorporates: #127
- resolves: #144
- resolves: #147

## Summary of Changes
- Added the **ThreadSafeRingBuffer**
  - with minimal unit tests for it
- Switch t [... cut]