ouster-lidar/ouster-ros #106: /os_cloud launches but does not publish pointcloud data

benripka, 2023-04-20:
I am attempting to run the os_sensor and os_cloud nodes via sensor.independent.launch.py  in order to convert my OS0 lidar packets into pointclouds. The os_sensor node seems to successfully produce lidar packet data, but the os_cloud node does not publish the clouds to any topic. Details are below any help would be greatly appreciated. Perhaps this is a bug?

- Ouster Sensor? OS0
- Ouster Firmware Version? 2.4
- ROS version/distro? foxy
- Operating System? Ubuntu Linux 20.04
- Machine Architecture? x86_64
- git branch: ros2-foxy
- robot: Clearpath Husky

The command I run is:

`ros2 launch ouster_ros sensor.independent.launch.py  viz:=false`

with my parameters.yaml set to:

`ouster/os_sensor:
  ros__parameters:
    sensor_hostname: '192.168.131.5'
    udp_dest: ''
    mtp_dest: ''
    mtp_main: false
    lidar_mode: ''
    timestamp_mode: ''
    udp_profile_lidar: ''
    metadata: ''
    lidar_port: 0
    imu_port: 0
    metadata: ''
ouster/os_cloud:
  ros__parameters:
    tf_prefix: 'ouster'
    timestamp_mode: 'TIME_FROM_ROS_TIME'`

The resulting output is:

`[INFO] [launch]: All log files can be found below /home/administrator/.ros/log/2023-04-19-20-35-19-061558-cpr-a200-0767-59255
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_sensor-1]: process started with pid [59267]
[INFO] [os_cloud-2]: process started with pid [59269]
[INFO] [os_image-3]: process started with pid [59271]
[os_cloud-2] [INFO] [1681950919.606634101] [ouster.os_cloud]: OusterCloud: node initialized!
[os_image-3] [INFO] [1681950919.615050137] [ouster.os_image]: OusterImage: node initialized!
[os_sensor-1] [WARN] [1681950919.810962946] [ouster.os_sensor]: lidar port set to zero, the client will assign a random port number!
[os_sensor-1] [WARN] [1681950919.811067090] [ouster.os_sensor]: imu port set to zero, the client will assign a random port number!
[os_sensor-1] [INFO] [1681950919.811085095] [ouster.os_sensor]: Will use automatic UDP destination
[os_sensor-1] [INFO] [1681950919.877258871] [ouster.os_sensor]: Sensor 192.168.131.5 configured successfully
[os_sensor-1] [INFO] [1681950919.877303996] [ouster.os_sensor]: Starting sensor 192.168.131.5 initialization...
[os_sensor-1] [2023-04-19 20:35:19.877] [ouster::sensor] [info] initializing sensor: 192.168.131.5 with lidar port/imu port: 0/0
[os_sensor-1] [2023-04-19 20:35:19.879] [ouster::sensor] [warning] udp socket(): Address family not supported by protocol
[os_sensor-1] [2023-04-19 20:35:19.882] [ouster::sensor] [warning] udp socket(): Address family not supported by protocol
[os_sensor-1] [INFO] [1681950939.887827258] [ouster.os_sensor]: ouster client version: 0.8.1+d730798-release
[os_sensor-1] product: OS-0-32-U1, sn: 992116000437, firmware rev: v2.4.0
[os_sensor-1] lidar mode: 1024x10, lidar udp profile: LEGACY
[os_sensor-1] [WARN] [1681950939.887894247] [LifecyclePublisher]: Trying to publish message on the topic '/ouster/metadata', but the publisher is not activated
[os_sensor-1] [INFO] [1681950939.887912160] [ouster.os_sensor]: No metadata file was specified, using: 192.168.131-metadata.json
[os_sensor-1] [INFO] [1681950939.888072929] [ouster.os_sensor]: Wrote sensor metadata to 192.168.131-metadata.json
[os_sensor-1] [INFO] [1681950939.889053475] [ouster.os_sensor]: reset service created
[os_sensor-1] [INFO] [1681950939.889966451] [ouster.os_sensor]: get_metadata service created
[os_sensor-1] [INFO] [1681950939.890796850] [ouster.os_sensor]: get_config service created
[os_sensor-1] [INFO] [1681950939.891652789] [ouster.os_sensor]: set_config service created
[INFO] [launch.user]: os_sensor activating...
`

After which is seems to just hang. The /ouster/lidar_packets topic is as-buzz with activity but there is  no /ouster/points topic.

Samahu, 2023-04-20:
@benripka Please watch for this hotfix https://github.com/ouster-lidar/ouster-ros/pull/107 which should address the issue you have reported.

Samahu, 2023-04-20:
The fix has been merged, please give it a try when you can.

benripka, 2023-04-20:
Works like a charm. Thanks for the quick turn around!

closed 2023-04-20 by benripka