ouster-lidar/ouster-ros #110: Pointcloud partially flickering (incomplete) in rviz.

batzor, 2023-04-24:
**Describe the bug**
PointCloud flickering in RVIZ. `ros2_ouster_driver` works without problem.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. ROS Foxy
2. Clone `ros2-foxy` branch and colcon build. 
3. `ros2 launch ouster_ros sensor.independent.launch.py`
4. params:
```
ouster/os_sensor:
  ros__parameters:
    sensor_hostname: '192.168.2.82'
    udp_dest: '192.168.2.81'
    mtp_dest: ''
    mtp_main: false
    lidar_mode: '512x10'
    timestamp_mode: 'TIME_FROM_ROS_TIME'
    udp_profile_lidar: 'LEGACY'
    metadata: ''
    lidar_port: 7502
    imu_port: 7503
ouster/os_cloud:
  ros__parameters:
    tf_prefix: ''
    timestamp_mode: ''
```

**Screenshots**
[image]

**Platform (please complete the following information):**
- Ouster Sensor? OS-0-64-U13
- Ouster Firmware Version? v2.4.0
- ROS version/distro? Foxy 
- Operating System? Ubuntu 20.04
- Machine Architecture? ARM
- git commit hash (if not the latest).

Samahu, 2023-04-24:
Do you connect to the sensor directly via an Ethernet cable or do you have a usb-ethernet converter in the middle?

In case you have used our ROS1 driver, have you observed similar behavior or you have only seen this issue when trying to integrate with the ROS2 driver?

I do suspect that this has to do with the ROS2 QoS settings. The driver currently uses SensorData QoS which uses "Best Effort" and has only 5 depth for the history. I do have ticket to make this configurable but if you are familiar with ROS2 and C++ you could try fiddling with them and see for example selecting "Reliable" QoS policy and/or increasing the history depth more than 5 helps.

rjwb1, 2023-04-25:
I'm also experiencing this issue on a pair of OS1-32 using a Jetson Orin with Ubuntu 20 and a 1Gbt Network switch. I have another system with this configuration that doesn't seem to have this issue. I am running ros 1 noetic.

Samahu, 2023-04-25:
Thank you for the input, and thank you @grischi for referencing the similar issue, I do have plans to change processing from timers to threads but I have considered this as an optimization, so it had less priority on my list. We'll try to address this ASAP.

grischi, 2023-04-25:
Hi @Samahu I have been able to resolve that issue based on your suggestion of QoS tuning. For the topic "lidar_packets", I have increased the history depth on publisher and subscriber side and that problem disappeared (the full cloud is now visible in rviz). Do you want me to send a PR?

grischi, 2023-04-25:
In my opinion, the "standard" user shall not be aware of the existence of a driver-internal topic like `lidar_packets`. I would therefore prefer not to have to configure QoS settings in a config file. In fact, I would favor a design that does not include such a topic at all, for performance reasons: The transport of messages of type PacketMsg.msg with a variable-length array cannot be done without creating a copy in cyclonedds/iceoryx (If I understood that correctly:  https://github.com/ros2/rmw_cyclonedds/blob/rolling/shared_memory_support.md#zero-copy).

Nosille, 2023-05-04:
I find the current method of publishing raw packets convenient for rosbag purposes.  Recording raw packets results in smaller bags, and we get to incorporate improvements made to the rest of the nodes on bag playback.  We can even vary some settings during playback sessions.

Samahu, 2023-05-05:
> I find the current method of publishing raw packets convenient for rosbag purposes. Recording raw packets results in smaller bags, and we get to incorporate improvements made to the rest of the nodes on bag playback. We can even vary some settings during playback sessions.

@Nosille the current plan is to make publishing raw packets optional, which would be off by default but enabled during record and replay scenarios. It is for the reasons that you have stated.

grischi, 2023-05-05:
Woudn't it be possible to cover the use-case of @Nosille with a network dump?

Samahu, 2023-05-05:
> Woudn't it be possible to cover the use-case of @Nosille with a network dump?

It would, but bag files are more convenient if you are already using ROS. Plus, recently we have started saving the sensor metadata into the bag file along the raw packets (since #102 and #103), which made it more convenient. With network dump the user would be responsible for maintaining and keeping record of the appropriate metadata file that match the recorded session (which very error prone).

Samahu, 2023-05-23:
> @Samahu I have this issue using ROS 1 on the master branch. Would it be possible to fix this issue there?

It is possible but I am prioritizing the fix for ros2 first then will back port it to ros1

PR #170 (2023-07-25): SW-5148: back port improvements in ros 2 to ros 1
## Related Issues & PRs
- original PR: #157 
- related: #146 
- related: #149
- related: #156 
- related: #162
- related: #163 
- resolves: #8 
- resolves: #33 
- resolves: #65 for custom frame names.
- resolves: #76
- resolves: #105 and #9 by implemented a switch between static and dynamic broadacasts of TF messages. 
- resolves: #110
- resolves: #137
- resolves: #144
- resolves:  #155
- incorporates: #127
- resolves: #144
- resolves: #147

## Summary of Changes
- Added the **ThreadSafeRingBuffer**
  - with minimal unit tests for it
- Switch t [... cut]
closed 2023-07-25 by Samahu