ouster-lidar/ouster-ros #126: Ouster os1 crashes in ros environment.

rupikaraj, 2023-05-08:
**Describe your question**
I am facing an issue while trying to launch ouster-os1 lidar from melodic ros environment. I have given below the screenshots of error. 

Kindly help me to sort out the issue.
Thank in advance.

**Platform**
- Ouster Sensor  OS-1
- ROS version Melodic
- Operating System : Linux

[image]
[image]

Samahu, 2023-05-08:
The driver is complaining that it can't resolve the hostname of the sensor, are you able to ping the device using the hostname form the machine that you are running on? try to see if you using the IP address works rather than the hostname?

Samahu, 2023-05-10:
@rupikaraj the problem is unlikely code related. For this type of issues you would get better assistance through [Ouster Customer Support](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/47/create/170), Closing the issue.

closed 2023-05-10 by Samahu