ouster-lidar/ouster-ros #137: Frame Rate Drop

mserour, 2023-05-11:
I have an OS1-64 running using ROS2 Foxy Driver, Ubuntu 20.04 that I received yesterday. My machine is quite powerful since it has the latest processor from Intel I9-13900K and I am using the standard launch file command to run the LIDAR. The LIDAR starts to consistently publish the data at 10 FPS, however, with time evolves, the FPS keeps dropping and there is some sort of skittering/discontinuity in the LIDAR visualization from RVIZ. 

When we tried to update the resolution of the output LIDAR image to 512x10, the rate was more consistent, however, it isn't that consistent over time (observed by hz the topic). Also one more observation, if we covered the LIDAR i.e. no points, the rate is consistent since there are no points. I believe it has something to do with the processing done to output the LIDAR image.

Any recommendations?

Thank you.

mserour, 2023-05-11:
I think it is a bit different because what is happening is some point cloud frames suddenly stop and then continue to publish again. It suddenly freezes on RVIZ and then it works normal again. Therefore, the frame rate drops when trying to ros topic hz the topic it self. Also, we didn't launch the node that publishes the LIDAR image and right now RVIZ is just visualizing the point cloud but the same issue happens that the point cloud freezes from time to time.

Samahu, 2023-05-11:
> The LIDAR starts to consistently publish the data at 10 FPS, however, with time evolves, the FPS keeps dropping and there is some sort of skittering/discontinuity in the LIDAR visualization from RVIZ.

How soon it takes to observe the drop in the frame rate?

mserour, 2023-05-11:
I think it is observed immediately, some frames freeze after operation. I am sorry the word of skittering or discontinuity is misleading and incorrect. It is different issue than 110. I think the problem right now is the frames freeze as visualized in RVIZ for some reason and then the average FPS is dropped because you lost some frames.

Samahu, 2023-05-11:
I carried out few tests around this issue, here are few findings:
- There is a noticeable drop in the frame rate when using the driver with ros-foxy
- It didn't matter whether I have used **FastRTP** or **CycloneDDs**, same behavior (I haven't tested other RMW since they are slightly more involved).
- This behavior is not observed when using the rolling distro on Ubuntu 20.04 machine.. meaning I would receive 10Hz on the `/ouster/points` and the rate is consistent. Also verified it visually, i.e there is not significant drop.
- I briefly tested the driver against humble at some point and don't recall having significant frame drop but another user reported poor performance on humble #63

In short I would say the complaint is valid but it is very distro/architecture/dds dependent. I would be addressing this and other concerns as part of a larger effort that I am working on. I will keep you updated on the issue. In the meantime you may consider experimenting with the rolling/humble if it is rather something that you can possibly consider.

PR #170 (2023-07-25): SW-5148: back port improvements in ros 2 to ros 1
## Related Issues & PRs
- original PR: #157 
- related: #146 
- related: #149
- related: #156 
- related: #162
- related: #163 
- resolves: #8 
- resolves: #33 
- resolves: #65 for custom frame names.
- resolves: #76
- resolves: #105 and #9 by implemented a switch between static and dynamic broadacasts of TF messages. 
- resolves: #110
- resolves: #137
- resolves: #144
- resolves:  #155
- incorporates: #127
- resolves: #144
- resolves: #147

## Summary of Changes
- Added the **ThreadSafeRingBuffer**
  - with minimal unit tests for it
- Switch t [... cut]
closed 2023-07-25 by Samahu