ouster-lidar/ouster-ros #144: corrupt scans

Janphr, 2023-05-25:
Hello!

I've been having problems lately with incomplete scans on both my OS1 (Rev 06 and D). 

I'm pretty sure I didn't have the problem in the past, but now, under some CPU load (e.g. with 'stress -c 4'), blocks of columns are missing.

Since I don't need 10hz and couldn't fix the problem by running different firmware and ouster-ros versions, I just filter out corrupt scans. These images should show the problem pretty well, with a larger and smaller area missing. These flicker over the entire 360 degrees and can be as large as 180 degrees. But since I only need the front part of the scan, I ignore the rest (the gray part) and only filter the scans that are corrupted in the front part.

[image]
[image]

- Ouster Sensor? 
OS1 Rev 06 and D
- Ouster Firmware Version? 
2.51 and 2.4
- ROS version/distro? 
noetic
- Operating System? 
linux 20.04
- Machine Architecture? 
i7-8565U (1.80GHz x 8)

Samahu, 2023-05-26:
Thanks for reporting the issue! Soon (early June) would should have an update that should reduce overall CPU utilization. I would expect it to help with your situation. I will keep you updated.

Janphr, 2023-06-15:
I forgot to mention, I have the same problem using Ouster Studio under Windows. 
Not as bad tho, but with no additional CPU load...
https://nextcloud.telematik-zentrum.de/index.php/s/rJYqg7TAPxAQsEL

With ouster-ros on linux, the flickering occurs more often and affects some larger areas. 
Sometimes several scans in a row are affected.

PR #170 (2023-07-25): SW-5148: back port improvements in ros 2 to ros 1
## Related Issues & PRs
- original PR: #157 
- related: #146 
- related: #149
- related: #156 
- related: #162
- related: #163 
- resolves: #8 
- resolves: #33 
- resolves: #65 for custom frame names.
- resolves: #76
- resolves: #105 and #9 by implemented a switch between static and dynamic broadacasts of TF messages. 
- resolves: #110
- resolves: #137
- resolves: #144
- resolves:  #155
- incorporates: #127
- resolves: #144
- resolves: #147

## Summary of Changes
- Added the **ThreadSafeRingBuffer**
  - with minimal unit tests for it
- Switch t [... cut]