ouster-lidar/ouster-ros #147: Visualisation Error

albait03, 2023-06-01:
**Describe your question**
Lately, I am having issues visualizing the 3D-Pointclouds in RVIZ. It starts with no issues, but after a few seconds the points just get rendered slowly in a circular motion. I attached a video of the issue below. 

This issue appeared after updating the firmware and using the latest ouster-ros release. I had everything working before with a 1 1/2 year old git release and old firmware (not sure about versions). 

On another workstation, I don't have any issues with the visualization at all. In ouster studio, everything works fine as well.

The only thing concerning I observed so far is the CPU utilization. When reproducing the error, it spikes to 100% on some cores (see attached Screenshot). In ouster studio, the utilization almost doesn't increase at all.

**Screenshots**
[image]
[image]

**Platform (please complete the following information):**
- Ouster Sensor? 
OS-0 and OS-1 (Rev D)
- Ouster Firmware Version? 
v2.5.1
- ROS version/distro? 
noetic
- Operating System? \[e.g. Linux, Windows, MacOS\]
Ubuntu 20.04
- Machine Architecture? 
x64 (Intel® Core™ i9-9900 CPU @ 3.10GHz × 16)

Samahu, 2023-06-01:
@albait03 thanks for reporting the issue. Did you get the chance to try the current driver with the previous FW? Would be helpful to isolate the issue.

I only see the circular patter when I have huge drop in packets (are using the same network configuration), the spike in CPU utilization could be simply a side effect of the packet drops (or possibly coming out of order) and this isn't playing nicely with the LidarScan assembly.

Waiting for your feedback.

albait03, 2023-06-08:
Thank you Samahu for the quick reply. 

> @albait03 thanks for reporting the issue. Did you get the chance to try the current driver with the previous FW? Would be helpful to isolate the issue.

Yes, I tested with the V2.4.0 Firmware and the issue still persists.

> I only see the circular patter when I have huge drop in packets (are using the same network configuration), the spike in CPU utilization could be simply a side effect of the packet drops (or possibly coming out of order) and this isn't playing nicely with the LidarScan assembly.

The network configuration is the same then it was before when it worked (with old driver/firmware), but it is indeed different to the simple workstation test. The LiDAR's are connected to a 10 GBit Networking Switch which is connected to the 10 GBit Networking Card in the System. I also tested the setup connecting the sensors directly to the system but the issue still persisted. Do you have any advice what I can do, to give you more specific information/how to check if i have packet drops? And does this even make sense concidering that Ouster Studio just works perfectly fine?

Kind regards

Samahu, 2023-06-08:
I do have an upcoming PR that should help with the overall performance and stability of the driver https://github.com/ouster-lidar/ouster-ros/pull/146 but this targets ROS2 branch for now then will back port certain portions to ROS1. In the meantime you can consult this [guide](https://static.ouster.dev/sensor-docs/image_route1/image_route2/appendix/analyzing-linux.html) and see if it helps out.

Samahu, 2023-07-25:
@albait03 I believe with the last update (package version 0.9.0) included in PR #170 your issue should be resolved. Make sure to use `driver.launch` instead of `sensor.launch` to obtain the improved performance and stability.

PR #170 (2023-07-25): SW-5148: back port improvements in ros 2 to ros 1
## Related Issues & PRs
- original PR: #157 
- related: #146 
- related: #149
- related: #156 
- related: #162
- related: #163 
- resolves: #8 
- resolves: #33 
- resolves: #65 for custom frame names.
- resolves: #76
- resolves: #105 and #9 by implemented a switch between static and dynamic broadacasts of TF messages. 
- resolves: #110
- resolves: #137
- resolves: #144
- resolves:  #155
- incorporates: #127
- resolves: #144
- resolves: #147

## Summary of Changes
- Added the **ThreadSafeRingBuffer**
  - with minimal unit tests for it
- Switch t [... cut]