ouster-lidar/ouster-ros #151: No message is publishing

zillur-av, 2023-06-16:
I followed the instructions properly and the launch file is running without any issues. I can visualize the pointcloud in Ouster Studio but in in rviz. Also nothing is shown on topic print like `ros2 topic echo ouster/points' too

This is the `sensor.launch.xml` file: 
```
  <arg name="ouster_ns" default="ouster"
    description="Override the default namespace of all ouster nodes"/>
  <arg name="sensor_hostname" default="os-992315000023.local/"
    description="hostname or IP in dotted decimal form of the sensor"/>
  <arg name="udp_dest" default="169.254.47.132"
    description="hostname or IP where the sensor will send data packets"/>
  <arg name="lidar_port" default="7502"
    description="port to which the sensor should send lidar data"/>
  <arg name="imu_port" default="7503"
    description="port to which the sensor should send imu data"/>
  <arg name="udp_profile_lidar" default="RNG19_RFL8_SIG16_NIR16_DUAL"
    description="lidar packet profile; possible values: {
    LEGACY,
    RNG19_RFL8_SIG16_NIR16_DUAL,
    RNG19_RFL8_SIG16_NIR16,
    RNG15_RFL8_NIR8
    }"/>
  <arg name="lidar_mode" default="1024x10"
    description="resolution and rate; possible values: {
    512x10,
    512x20,
    1024x10,
    1024x20,
    2048x10,
    4096x5
    }"/>
  <arg name="timestamp_mode" default=" TIME_FROM_ROS_TIME"
    description="method used to timestamp measurements; possible values: {
    TIME_FROM_INTERNAL_OSC,
    TIME_FROM_SYNC_PULSE_IN,
    TIME_FROM_PTP_1588,
    TIME_FROM_ROS_TIME
    }"/>
  <arg name="metadata" default=""
    description="path to write metadata file when receiving sensor data"/>
  <arg name="viz" default="true"
    description="whether to run a rviz"/>
  <arg name="rviz_config" default="$(find-pkg-share ouster_ros)/config/viz.rviz"
    description="optional rviz config file"/>
```
This is the `parameters.yaml` file:
```
ouster/os_sensor:
  ros__parameters:
    sensor_hostname: 'os-992315000023.local'
    udp_dest: ''
    mtp_dest: ''
    mtp_main: false
    lidar_mode: ''
    timestamp_mode: ''
    udp_profile_lidar: ''
    metadata: ''
    lidar_port: 0
    imu_port: 0
ouster/os_cloud:
  ros__parameters:
    sensor_frame: 'os_sensor'
    lidar_frame: 'os_lidar'
    imu_frame: 'os_imu'
    timestamp_mode: 'TIME_FROM_ROS_TIME'  # this value needs to match os_sensor/timestamp_mode
```

This is the web interface:
[image]

THis is the terminal output:
[... 38 lines cut ...]
[bash-2] Transitioning successful
[bash-3] Transitioning successful
[INFO] [bash-3]: process has finished cleanly [pid 6114]
[INFO] [bash-2]: process has finished cleanly [pid 6112]
[bash-4] [INFO] [1686958793.156411328] [rviz2]: Stereo is NOT SUPPORTED
[bash-4] [INFO] [1686958793.156480377] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[bash-4] [INFO] [1686958793.193678662] [rviz2]: Stereo is NOT SUPPORTED
[component_container_mt-1] [WARN] [1686958793.273401379] [ouster.os_image]: New subscription discovered on topic '/ouster/nearir_image', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[component_container_mt-1] [WARN] [1686958793.274454862] [ouster.os_image]: New subscription discovered on topic '/ouster/range_image', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[component_container_mt-1] [WARN] [1686958793.276034180] [ouster.os_cloud]: New subscription discovered on topic '/ouster/points', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[component_container_mt-1] [WARN] [1686958793.277021623] [ouster.os_image]: New subscription discovered on topic '/ouster/reflec_image', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[component_container_mt-1] [WARN] [1686958793.278114512] [ouster.os_image]: New subscription discovered on topic '/ouster/signal_image', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[bash-4] [INFO] [1686958793.326565288] [rviz2]: Stereo is NOT SUPPORTED
[component_container_mt-1] [WARN] [1686958810.024634026] [ouster.os_cloud]: New subscription discovered on topic '/ouster/points', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[bash-4] [WARN] [1686958810.032629958] [ouster.rviz2]: New publisher discovered on topic '/ouster/points', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[component_container_mt-1] [WARN] [1686958814.138908595] [ouster.os_cloud]: New subscription discovered on topic '/ouster/points', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[bash-4] [WARN] [1686958814.145534448] [ouster.rviz2]: New publisher discovered on topic '/ouster/points', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
```

- Ouster Sensor: OS-2
- Ouster Firmware Version: 2.5.1
- ROS version/distro: humble
- Operating System: ubuntu 22.04
- Machine Architecture: x64,
- git commit hash (if not the latest).

Samahu, 2023-06-19:
From the logs, I can see that you connected successfully to the sensor. What I don't get is why are you getting: "requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY". Are you using a different rviz file than the default one? This mismatch would prevent RVIZ from displaying the point cloud.

With regard to `ros2 topic echo ouster/points` try prefixing the the topic name with `/` -> `ros2 topic echo /ouster/points`

zillur-av, 2023-06-19:
> From the logs, I can see that you connected successfully to the sensor. What I don't get is why are you getting: "requesting incompatible QoS. No messages will be sent to it. Last incompatible [...]

Thanks for your reply. I did not change anything to the provided rviz file. Not sure what happened, but it is working now.

Samahu, 2023-06-19:
Great! I guess we can close this issue then.

closed 2023-06-19 by Samahu

chadrs2, 2024-09-11:
I'm getting the same issue in ros2 when trying to display the point clouds. It only lets me display the point cloud when I change the PointCloud2 settings in rviz from 'Reliable' to 'System Default'