ouster-lidar/ouster-ros #158: Build Galactic Ouster Driver, But Do Not Receive Point Cloud or Signal Image

gauravkuppa, 2023-06-30:
**Describe the bug**
I build the branch `galactic_build_check` on my galactic ROS running on Xavier. Everything builds fine, but when I run the launch file, I do not see the corresponding topics getting populated with messages.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. source ros environment
2. turn off viz (viz=False)
3. colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-select ouster_ros ouster_srvs ouster_msgs
4. . install/setup.bash
5. ros2 launch ouster_ros sensor.independent.launch.xml sensor_hostname:=XXX.XXX.XXX udp_dest:=XXX.XXX.XXX  lidar_port:=<> imu_port:=<>

**Screenshots**
If applicable, add screenshots to help explain your problem.
```[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_sensor-1]: process started with pid [15428]
[INFO] [os_cloud-2]: process started with pid [15430]
[INFO] [os_image-3]: process started with pid [15431]
[INFO] [ros2-4]: process started with pid [15432]
[INFO] [ros2-5]: process started with pid [15433]
[os_image-3] [INFO] [1688167502.374745072] [ouster.os_image]: OusterImage: node initialized!
[os_cloud-2] [INFO] [1688167502.455768140] [ouster.os_cloud]: OusterCloud: node initialized!
[os_sensor-1] [INFO] [1688167504.043047398] [ouster.os_sensor]: Will send UDP data to 192.168.168.234
[os_sensor-1] [INFO] [1688167504.081565232] [ouster.os_sensor]: Sensor 192.168.168.209 configured successfully
[os_sensor-1] [INFO] [1688167504.081852391] [ouster.os_sensor]: Starting sensor 192.168.168.209 initialization...
[os_sensor-1] [2023-06-30 19:25:04.082] [ouster::sensor] [info] initializing sensor: 192.168.168.209 with lidar port/imu port: 47044/40379
[os_sensor-1] [INFO] [1688167505.194583726] [ouster.os_sensor]: ouster client version: 0.8.1+d730798-release
[os_sensor-1] product: OS-1-128, sn: 122314002197, firmware rev: v3.0.1
[os_sensor-1] lidar mode: 1024x10, lidar udp profile: RNG19_RFL8_SIG16_NIR16
[os_sensor-1] [INFO] [1688167505.195050739] [ouster.os_sensor]: No metadata file was specified, using: 192.168.168-metadata.json
[os_sensor-1] [INFO] [1688167505.195697030] [ouster.os_sensor]: Wrote sensor metadata to 192.168.168-metadata.json
[os_sensor-1] [INFO] [1688167505.198127335] [ouster.os_sensor]: reset service created
[os_sensor-1] [INFO] [1688167505.200293042] [ouster.os_sensor]: get_metadata service created
[os_sensor-1] [INFO] [1688167505.202664686] [ouster.os_sensor]: get_config service created
[os_sensor-1] [INFO] [1688167505.207168146] [ouster.os_sensor]: set_config service created
[os_image-3] [INFO] [1688167505.195437746] [ouster.os_image]: OusterImage: retrieved new sensor metadata!
[os_cloud-2] [INFO] [1688167505.195587934] [ouster.os_cloud]: OusterCloud: retrieved new sensor metadata!
[os_image-3] [WARN] [1688167505.237205690] [ouster.os_image]: New subscription discovered on topic '/ouster/range_image', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[os_image-3] [WARN] [1688167505.237467215] [ouster.os_image]: New subscription discovered on topic '/ouster/reflec_image', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[ros2-5] Transitioning successful
[ros2-4] Transitioning successful
[INFO] [ros2-5]: process has finished cleanly [pid 15433]
[INFO] [ros2-4]: process has finished cleanly [pid 15432]
```
I do not see any topics updated in 

```
/ouster/imu
/ouster/imu_packets
/ouster/lidar_packets
/ouster/nearir_image
/ouster/points
/ouster/range_image
/ouster/reflec_image
/ouster/signal_image
```
except /ouster/os_sensor/transition_event

```
ros2 topic echo /ouster/os_sensor/transition_event
timestamp: 0
transition:
  id: 0
  label: ''
start_state:
  id: 1
  label: unconfigured
goal_state:
  id: 10
  label: configuring
---
timestamp: 0
transition:
  id: 0
  label: ''
start_state:
  id: 1
  label: unconfigured
goal_state:
  id: 10
  label: configuring
---
timestamp: 0
transition:
  id: 0
  label: ''
start_state:
  id: 10
  label: configuring
goal_state:
  id: 2
  label: inactive
---
timestamp: 0
transition:
  id: 0
  label: ''
start_state:
  id: 2
  label: inactive
goal_state:
  id: 13
  label: activating
---
timestamp: 0
transition:
  id: 0
  label: ''
start_state:
  id: 13
  label: activating
goal_state:
  id: 3
  label: active
---
```
**Platform (please complete the following information):**
- Ouster Sensor? OS-1 128
- Ouster Firmware Version? 3.0.1
- ROS version/distro? Galactic
- Operating System? NVIDIA Xavier
- Machine Architecture? aarch64
- git commit hash (if not the latest). a112dc1

closed 2023-07-01 by gauravkuppa

Samahu, 2023-07-03:
@gauravkuppa I noticed that you closed the issue, did you manage to get the point cloud from the driver?

Samahu, 2023-07-03:
also note that you don't need to be using the `galactic_build_check` branch which isn't going to be updated. Instead you should use the `ros2-foxy` branch.