ouster-lidar/ouster-ros #159: Bug on ROS2 branch after the last MR.

abourbou, 2023-07-03:
**Describe the bug**
Hi, I am using the ROS2 branch with our OS-1 64 and I have problem today about the live usage of your launch file.

**To Reproduce**
I git clone the repo on the ros2 branch and I built it with
```bash
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-up-to ouster_ros
source install/setup.zsh
```
after connecting my sensor and getting the IP address on http://os1-99numbers.local/
I tried to launch your launch file :
```bash
ros2 launch ouster_ros sensor.launch.xml    \
    sensor_hostname:="ip_address"                      
[INFO] [launch]: All log files can be found below /home/abourbou/.ros/log/2023-07-03-15-16-06-944852-abourbou-XPS-15-9560-49833
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_driver-1]: process started with pid [49834]
[INFO] [bash-2]: process started with pid [49836]
[INFO] [bash-3]: process started with pid [49838]
[INFO] [bash-4]: process started with pid [49841]
[bash-2] Node not found
[ERROR] [bash-2]: process has died [pid 49836, exit code 1, cmd 'bash -c sleep 0; $0 $@ /opt/ros/humble/bin/ros2 lifecycle set /ouster/os_sensor configure'].
[bash-3] Node not found
[ERROR] [bash-3]: process has died [pid 49838, exit code 1, cmd 'bash -c sleep 1; $0 $@ /opt/ros/humble/bin/ros2 lifecycle set /ouster/os_sensor activate'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ERROR] [bash-4]: process has died [pid 49841, exit code -2, cmd 'bash -c sleep 5; $0 $@ /opt/ros/humble/lib/rviz2/rviz2 -d /home/abourbou/Desktop/Projects/ws_ouster_driver/install/ouster_ros/share/ouster_ros/config/viz.rviz --ros-args -r __node:=rviz2 -r __ns:=/ouster'].
[os_driver-1] [INFO] [1688390170.640257017] [rclcpp]: signal_handler(signum=2)
```

I tried using the python launch file 
```bash
ros2 launch ouster_ros sensor.composite.launch.py    \    
    params_files:="absolute/path/to/ouster_params.yaml"
```
 with inside : sensor_hostname = "ip_address" but it doesn't seems to detect the host_name
 ```bash
[component_container_mt-2] [ERROR] [1688390419.552914044] [ouster.os_sensor]: Must specify a sensor hostname
[component_container_mt-2] [ERROR] [1688390419.552983238] [ouster.os_sensor]: exception thrown while configuring the sensor, details: Must specify a sensor hostname
[component_container_mt-2] [WARN] [1688390419.553014213] []: Error occurred while doing error handling.
[os_sensor configure-3] Transitioning failed
[INFO] [os_sensor activate-4]: process started with pid [50169]
[INFO] [os_sensor configure-3]: process has finished cleanly [pid 50093]
[os_sensor activate-4] Unknown transition requested, available ones are:
[ERROR] [os_sensor activate-4]: process has died [pid 50169, exit code 1, cmd '/opt/ros/humble/bin/ros2 lifecycle set ouster/os_sensor activate
```

**Platform**
- Ouster Sensor : OS-1 64
- Ouster Firmware Version : v2.3.0
- ROS version : humble
- Operating System : Ubuntu 22

** Resolve **
By going back the commit of may : 29f8be331546a540238342c7e99a4b6c1e471eb7
this command works
```bash
ros2 launch ouster_ros sensor.launch.xml    \
    sensor_hostname:="169.254.47.236"
```
but the python launchfile doesn't with the same error message than previously.

Samahu, 2023-07-03:
@abourbou When using the xml launch file they lack support for event based transition and use time based to activate the node. It generally works just fine but in certain cases for example when running within a docker container that may not work very well. In this case I would recommend using the python launch files.

For the python based launch file there was a typo on my side where I misspelled the `params_file` and wrote instead `params_files` with an extra `s` character at the end. I do apologies for that and will remove the typo immediately. Could you try correcting your launch command to the following and try again:
```bash
ros2 launch ouster_ros sensor.composite.launch.py    \    
    params_file:="absolute/path/to/ouster_params.yaml"
```

Samahu, 2023-07-03:
@abourbou I have updated the `README.`md to correct the typo. Please let me know if the updated command works for you so we can mark this issue as closed.

abourbou, 2023-07-04:
Hi firstly thank you for your prompt reply,
 it works by modifying the default parameter yaml file but the default one is not the same as the one noted on the README : https://github.com/ouster-lidar/ouster-ros/tree/ros2#sensor-mode. The code indicate https://github.com/ouster-lidar/ouster-ros/blob/ros2/ouster-ros/launch/sensor.composite.launch.py#L25
 
By modifying os_sensor_cloud_image_params.yaml with my IP adress it works.
Creating a parameter file by copying os_sensor_cloud_image_params.yaml and replace by my own parameters also works.
Thank you for the help

closed 2023-07-04 by abourbou

Samahu, 2023-07-04:
`driver.launch.py` uses the `driver_params.yaml`, `sensor.composite.launch.py` on the other hand uses `os_sensor_cloud_image_params.yaml`. I admit that our main readme is lacking this detail. You are free to choose between `driver.launch.py` or `sensor.composite.launch.py` but `driver.launch.py` has less overhead overall and is more stable in terms of Frame Rate.