ouster-lidar/ouster-ros #160: Publishing a 2D laser scan

tanelikor, 2023-07-04:
**Is your feature request related to a problem? Please describe.**
One of the features that we have found helpful in the community driver is the possibility to get 2D LaserScan from the lidar. This is needed especially when using the sensor for the purposes of 2D SLAM etc.

Are there plans to add such a feature in this official driver, in particular the ROS 2 version? I also wouldn't mind contributing to this feature, though I likely won't have time for it at least for a month or so.

**Describe the solution you'd like**
Build a 2D scan from one or both of the middle scan rings of the lidar point cloud and publish it in a LaserScan message.

**Describe alternatives you've considered**
I've briefly thought of simply constructing the scan from the point cloud (just find the middle ring(s) and compute the distances from them), but it could of course be more efficient to do it utilizing the sdk if possible.

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
- Ouster Sensor? N/A
- Ouster Firmware Version? N/A
- ROS version/distro? N/A
- Operating System? N/A
- Machine Architecture? N/A

Samahu, 2023-07-04:
@tanelikor which ROS version are you targeting? We have recently added this feature to the ROS2 driver (but not the ROS1) as part of the PRs: {#146, #149}. You can utilize this message by subscribing to `/ouster/scan` topic which is only enabled if you set the `SCAN` flag for the `proc_mask` parameter: https://github.com/ouster-lidar/ouster-ros/blob/6c9b3f514c7e4c0c8e1240c47c1c2ecbd3c7a365/ouster-ros/config/driver_params.yaml#L58.

We have added support for all the essential features that the ROS2 community driver for Ouster had but the official one lacked as part of these two referenced PRs. This is an effort to bridge the gap and have a single solution for the entire Ouster/ROS community.

tanelikor, 2023-07-05:
@Samahu Oh, my apologies: I completely missed this yesterday when browsing through the recent updates, and the testing I did was about 3 weeks ago when this update wasn't there yet. This is my mistake, sorry for the inconvenience. 

We are currently targeting ROS 2 Humble. Since the feature is now added (many thanks for that!), I think this ticket can be closed. I will give the scan a test when I have time.

Samahu, 2023-07-05:
> @Samahu Oh, my apologies: I completely missed this yesterday when browsing through the recent updates, and the testing I did was about 3 weeks ago when this update wasn't there yet. This is my mistake, sorry for the inconvenience.

No worries! I am always happy to see that we get ahead on feature requests (every once in a while).

I will close this ticket, feel free to re-open the ticket in case it does not work for you. also note that we have `scan_ring` parameter which you can use along the `SCAN` to select which ring you want to use without having to modify the code. For more details refer to: https://github.com/ouster-lidar/ouster-ros/blob/6c9b3f514c7e4c0c8e1240c47c1c2ecbd3c7a365/ouster-ros/config/driver_params.yaml#L59-L62.

closed 2023-07-05 by Samahu

tanelikor, 2023-07-06:
@Samahu Thank you! We'll give it a try and report back in case we see any problems.