ouster-lidar/ouster-ros #161: ROS1 driver setup might hang due to blocking OnInit() if launch file sleeps are too low

skohlbr, 2023-07-06:
**Describe the bug**
When loading the Ouster nodelets into an existing nodelet manager, I noticed that sometimes the LIDAR does not start streaming. The other thing I noticed is that running `rosservice call /existing_nodelet_manager/list` doesn't return when this happens, indicating that very likely some nodelet is blocking in `OnInit()`. It appears the issue can be made to happen more frequently by reducing the sleep times in the `launch-prefix` entries (i.e. `launch-prefix="bash -c 'sleep 6; $0 $@' "`). I'd be interested why those weird sleeps are there in the first place and strongly suggest to make things work without them as this appears very brittle (due to system load and other factors might be impacting when a sleep duration is "enough" or not).

**To Reproduce**
Assume there is a existing `existing_nodelet_manager` getting started. Simultaneously start below launch file. When the sleeps are "too short" the LIDAR might not come up correctly. I managed to make it work by doubling the sleep times, but as mentioned above this appears quite brittle.

**Platform (please complete the following information):**
- OS0-32 Rev.6
- 2.51
- ROS melodic
- Ubuntu 18.04
- Arm64/Jetson

```xml
<?xml version="1.0"?>
<launch>
  <arg name="manager" default="existing_nodelet_manager" />
  <arg name="ouster_ns" default="ouster" doc="Override the default namespace of all ouster nodes"/>
  <arg name="sensor_hostname" default="192.168.45.50" doc="hostname or IP in dotted decimal form of the sensor" />
  <arg name="udp_dest" default="192.168.45.1" doc="hostname or IP where the sensor will send data packets" />
  <arg name="lidar_port" default="55500" doc="port to which the sensor should send lidar data" />
  <arg name="imu_port" default="55501" doc="port to which the sensor should send imu data" />
  <arg name="metadata" default="/home/robot_xy/logs/test.log" doc="override default metadata file for replays" />
  <arg name="image" default="true" doc="publish range/intensity/ambient image topic" />
  <arg name="tf_prefix" default="" doc="namespace for tf transforms" />
  
  <arg name="udp_profile_lidar" default=" " doc="lidar packet profile; possible values: {
    LEGACY,
    RNG19_RFL8_SIG16_NIR16_DUAL,
    RNG19_RFL8_SIG16_NIR16,
    RNG15_RFL8_NIR8
    }"/>
  <arg name="lidar_mode" default=" " doc="resolution and rate; possible values: {
    512x10,
    512x20,
    1024x10,
    1024x20,
    2048x10,
    4096x5
    }"/>
  <arg name="timestamp_mode" default=" " doc="method used to timestamp measurements; possible values: {
    TIME_FROM_INTERNAL_OSC,
    TIME_FROM_SYNC_PULSE_IN,
    TIME_FROM_PTP_1588,
    TIME_FROM_ROS_TIME
    }"/>
    
  <group ns="$(arg ouster_ns)">
    <node pkg="nodelet" type="nodelet" name="os_node"
      output="screen"
      launch-prefix="bash -c 'sleep 3; $0 $@' "
      args="load nodelets_os/OusterSensor /$(arg manager)">
      <param name="~/sensor_hostname" type="str" value="$(arg sensor_hostname)"/>
      <param name="~/udp_dest" type="str" value="$(arg udp_dest)"/>
      <param name="~/lidar_port" type="int" value="$(arg lidar_port)"/>
      <param name="~/imu_port" type="int" value="$(arg imu_port)"/>
      <param name="~/udp_profile_lidar" type="str" value="$(arg udp_profile_lidar)"/>
      <param name="~/lidar_mode" type="str" value="$(arg lidar_mode)"/>
      <param name="~/timestamp_mode" type="str" value="$(arg timestamp_mode)"/>
      <param name="~/metadata" type="str" value="$(arg metadata)"/>
    </node>
    
    <node pkg="nodelet" type="nodelet" name="os_cloud_node"
      output="screen"
      launch-prefix="bash -c 'sleep 4; $0 $@' "
      args="load nodelets_os/OusterCloud /$(arg manager)">
      <param name="~/tf_prefix" type="str" value="$(arg tf_prefix)"/>
      <param name="~/timestamp_mode" type="str" value="$(arg timestamp_mode)"/>
    </node>
    
    <node if="$(arg image)" pkg="nodelet" type="nodelet" name="img_node"
      output="screen"
      launch-prefix="bash -c 'sleep 4; $0 $@' "
      args="load nodelets_os/OusterImage /$(arg manager)">
    </node>
  </group>
  
</launch>

.```

skohlbr, 2023-09-10:
Ok, just saw that those sleeps have been removed in https://github.com/ouster-lidar/ouster-ros/commit/4d6476e078ca3d2254131e62efeb1e126816cca1, will check with current version.

Samahu, 2023-09-11:
@skohlbr  Yes, please let me know if these changes resolve this issue.