ouster-lidar/ouster-ros #165: Software or hardware related issue ?

AxelCORNILLEAU, 2023-07-12:
roslaunch ouster_ros os0.launch (properly set up, used to work)
```
... logging to /root/.ros/log/d2350fb0-2d94-11ed-9a24-48b02d3c814f/roslaunch-044bc8ee3883-153352.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://044bc8ee3883:45717/

SUMMARY
========

PARAMETERS
 * /os0/os_cloud_node/tf_prefix: os0
 * /os0/os_cloud_node/timestamp_mode: TIME_FROM_ROS_TIME
 * /os0/os_node/imu_port: 7503
 * /os0/os_node/lidar_mode: 2048x10
 * /os0/os_node/lidar_port: 7502
 * /os0/os_node/metadata: /opt/ros/noetic/s...
 * /os0/os_node/sensor_hostname: 169.254.113.23
 * /os0/os_node/timestamp_mode: TIME_FROM_ROS_TIME
 * /os0/os_node/udp_dest: 
 * /os0/os_node/udp_profile_lidar: 
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)
  /os0/
    img_node (nodelet/nodelet)
    os_cloud_node (nodelet/nodelet)
    os_node (nodelet/nodelet)
    os_nodelet_mgr (nodelet/nodelet)

ROS_MASTER_URI=http://169.254.1.10:11311

process[os0/os_nodelet_mgr-1]: started with pid [153378]
process[os0/os_node-2]: started with pid [153380]
process[os0/os_cloud_node-3]: started with pid [153382]
process[os0/img_node-4]: started with pid [153384]
process[relay_lidar-5]: started with pid [153386]
process[relay_imu-6]: started with pid [153387]
[ INFO] [1689023607.459188480]: Initializing nodelet with 12 worker threads.
[ INFO] [1689023608.438053376]: Loading nodelet /os0/os_node of type nodelets_os/OusterSensor to manager os_nodelet_mgr with the following remappings:
[ INFO] [1689023608.541361312]: TIME_FROM_ROS_TIME timestamp mode specified. IMU and pointcloud messages will use ros time
[ INFO] [1689023608.541421472]: Will use automatic UDP destination
[ INFO] [1689023608.859731936]: Sensor 169.254.113.23 was configured successfully
[ INFO] [1689023608.859794912]: Starting sensor 169.254.113.23 initialization...
[2023-07-10 21:13:28.859] [ouster::sensor] [info] initializing sensor: 169.254.113.23 with ports: 7502/7503
[ INFO] [1689023609.452166976]: Loading nodelet /os0/os_cloud_node of type nodelets_os/OusterCloud to manager os_nodelet_mgr with the following remappings:
[ INFO] [1689023609.462283264]: Loading nodelet /os0/img_node of type nodelets_os/OusterImage to manager os_nodelet_mgr with the following remappings:
[2023-07-10 21:13:43.148] [ouster::sensor] [warning] The SDK will soon output the non-legacy metadata format by default.  If you parse the metadata directly instead of using the SDK (which will continue to read both legacy and non-legacy formats), please be advised that on the next release you will either have to update your parsing or specify legacy_format = true to the get_metadata function.
[ INFO] [1689023623.273617632]: ouster client version: 0.7.1+cdc5605-release
product: OS-0-32-BH13, sn: 122217002133, firmware rev: v2.4.0
lidar mode: 2048x10, lidar udp profile: LEGACY
[ INFO] [1689023623.282656992]: Wrote metadata to /opt/ros/noetic/share/ros/share/ouster_ros/launch/os0.json
[ INFO] [1689023623.287337184]: get_metadata service created
[ INFO] [1689023623.291652960]: get_config service created
[ INFO] [1689023623.296851264]: set_config service created
[ INFO] [1689023624.307988544]: OusterCloud: retrieved sensor metadata!
[ INFO] [1689023625.309184576]: OusterImage: retrieved sensor metadata!
[ INFO] [1689023636.391195872]: Bond broken, exiting
================================================================================REQUIRED process [os0/os_node-2] has died!
process has finished cleanly
log file: /root/.ros/log/d2350fb0-2d94-11ed-9a24-48b02d3c814f/os0-os_node-2*.log
Initiating shutdown!
================================================================================
[relay_imu-6] killing on exit
[relay_lidar-5] killing on exit
[os0/img_node-4] killing on exit
[os0/os_cloud_node-3] killing on exit
[os0/os_node-2] killing on exit
[os0/os_nodelet_mgr-1] killing on exit
[ INFO] [1689023636.848486816]: Unloading nodelet /os0/img_node from manager os_nodelet_mgr
[ INFO] [1689023636.848574816]: Unloading nodelet /os0/os_cloud_node from manager os_nodelet_mgr
shutting down processing monitor...
... shutting down processing monitor complete
done

```
This issue appeared suddenly in both my os0 and os1 ouster lidar today, ROS noetic, ubuntu 20.04. What should I look into ? hardware ? software ? network ? I'm a bit lost and the debug ain't specific. Is that somekind of hearthbeat that stopped ? Network delay ?
Thanks for the help ;)
Axl

Samahu, 2023-07-12:
Your log says: "[ INFO] [1689023636.391195872]: Bond broken, exiting" which could indicate that a node or a nodelet has crashed or some of your nodes are out of sync. Do you run all of your nodes on the same machine?

AxelCORNILLEAU, 2023-07-13:
Hello Samahu,
Yes I do. Just the rosmaster is on another device.
ROS_MASTER_URI=http://169.254.1.20:11311 while I am on 169.254.1.20.

Samahu, 2023-07-17:
Until you fix the time sync between the different machines you can set the  **no_bond** launch file argument to **true**. This tells ROS not to use a bond when creating the driver nodes/notelets and ignore the heartbeat check.

AxelCORNILLEAU, 2023-07-17:
Okay thanks for the tip. the bug might be related to docker port. I'm on vacation I'll keep you updated when I'll look deeper into it ;)

AxelCORNILLEAU, 2023-08-01:
Turns out the ntp service was not installed on the computer. Thought it was and therefore looked elsewhere, my bad.

Samahu, 2023-08-01:
Yeah, you need to have the ntp service for sure.

closed 2023-08-01 by Samahu