ouster-lidar/ouster-ros #166: Running ouster driver in docker - "Must specify a sensor hostname"

LukasBergs, 2023-07-13:
Hi!

**Describe the bug**
I built the ouster driver within in a docker image but when I start the node I get the error message "Must specify a sensor hostname" even though it is set in the driver_params.yaml. I am sure that the launch file uses the correct params file. In https://github.com/ouster-lidar/ouster-ros/issues/159 I saw that you mentioned that there might be issues using docker. How can I fix this?

**How to reproduce**
- Download the files that I use to build the image and run the docker container here: https://rwth-aachen.sciebo.de/s/eollHDl2k0LJfbZ
- Install tmux with `apt-get install tmux`
- Execute `bash docker_build.sh`
- Run the docker container using `bash docker_run/run_ouster_driver.sh` (eventually you have to make some changes here, especially the variables HOST_IP and HOST_NAME

**Video**
You can also find a video of the problem in the uploaded folder. 

**Platform:**
- Ouster Sensor: OS1-128
- Ouster Firmware Version: newest
- ROS version: humble
- Operating System: Ubuntu 22.04
- Machine Architecture: x64
- git branch: ros2

I would be very grateful for any help!
Lukas

Samahu, 2023-07-17:
Hi @LukasBergs there is no issues using docker to run the ouster driver. I do suspect that the issue you are observing is a matter of wrong path supplied to the params_file. Is `/overlay_ws/src/custom_config_launch_files/config/driver_params.yaml` a valid path? Is it accessible within the container? I don't see that you mapped the path to `/overlay_ws` when you launch docker so how is it possible that you pass this file to the container?

LukasBergs, 2023-07-17:
Hi @Samahu,
Thank you for your response! 

Within the docker build process I copy the content of the overlay workspace to the root directory of the image. In the attached video you can see that the node receives the params yaml file. If I make changes in it (e.g. syntax errors) I get an error message when starting the node. Nevertheless, it does not seem to receive the sensor_hostname parameter. 

Do you have any other guesses where the problem might be coming from?

Best
Lukas

Samahu, 2023-07-17:
@LukasBergs when opting for a different ROS namespace for a node you config params file should reflect that too. Since you renamed the namespace you have to edit the config params files and prefix the node name with the modified namespace.

LukasBergs, 2023-07-18:
Thanks a lot @Samahu !
I was not aware of that. After adjusting the namespace, the parameter is recognized by the ouster driver.

closed 2023-07-18 by LukasBergs