ouster-lidar/ouster-ros #169: ROS2 Windows support?

headlee, 2023-07-19:
**Describe your question**
I was wondering if anyone has had luck building the `ros2` branch on Windows. I'm running into some CMake configuration and then MSVC compilation issues...I can get through some but can't seem to get everything to build together.

**Platform (please complete the following information):**
- ROS version/distro? Humble
- Operating System? Windows 10 (Visual Studio 2019)
- Machine Architecture? x64
- git commit hash (if not the latest).

Samahu, 2023-07-19:
Hi @headlee, I spent very little time trying to set up and build the driver under the Windows platform but I didn't going through it fully. However, I want to mention that we will soon (within this or next month) integrate the official ouster driver within ros2_drivers build farms. As part of this task we will definitely have the opportunity to checkout out and address any issues when building or deploying the driver to Windows platform.

headlee, 2023-07-24:
I was able to get everything built but haven't had a chance to test. I'm not sure if the changes are best practice (or if they break the Linux build) so I don't want to make a MR, but here is a diff in case it helps you/anyone else.

```patch
diff --git a/ouster-ros/CMakeLists.txt b/ouster-ros/CMakeLists.txt
index d91ea20..8962a8b 100644
--- a/ouster-ros/CMakeLists.txt
+++ b/ouster-ros/CMakeLists.txt
@@ -24,7 +24,12 @@ find_package(pcl_conversions REQUIRED)
 find_package(tf2_eigen REQUIRED)
 
 # ==== Options ====
-add_compile_options(-Wall -Wextra)
+if(MSVC)
+  add_compile_options(/W2 /wd4996)
+  add_compile_definitions(NOMINMAX _USE_MATH_DEFINES WIN32_LEAN_AND_MEAN)
+else()
+  add_compile_options(-Wall -Wextra)
+endif()
 if(NOT DEFINED CMAKE_CXX_STANDARD)
   set(CMAKE_CXX_STANDARD 17)
   set(CMAKE_CXX_STANDARD_REQUIRED ON)
@@ -56,7 +61,7 @@ include_directories(${_ouster_ros_INCLUDE_DIRS})
 # use only MPL-licensed parts of eigen
 add_definitions(-DEIGEN_MPL2_ONLY)
 
-add_library(ouster_ros_library SHARED
+add_library(ouster_ros_library SHARED OBJECT
   src/os_ros.cpp
 )
 
diff --git a/ouster-ros/test/ring_buffer_test.cpp b/ouster-ros/test/ring_buffer_test.cpp
index 8d8bd7f..1ef01fb 100644
--- a/ouster-ros/test/ring_buffer_test.cpp
+++ b/ouster-ros/test/ring_buffer_test.cpp
@@ -102,7 +102,7 @@ TEST_F(ThreadSafeRingBufferTest, ReadWriteToBuffer) {
     EXPECT_TRUE(buffer->empty());
     EXPECT_FALSE(buffer->full());
 
-    std::thread producer([this, &source]() {
+    std::thread producer([this, &source, TOTAL_ITEMS]() {
         for (int i = 0; i < TOTAL_ITEMS; ++i) {
             buffer->write([i, &source](uint8_t* buffer){
                 std::memcpy(buffer, &source[i][0], ITEM_SIZE);
@@ -110,7 +110,7 @@ TEST_F(ThreadSafeRingBufferTest, ReadWriteToBuffer) {
         }
     });
 
-    std::thread consumer([this, &target]() {
+    std::thread consumer([this, &target, TOTAL_ITEMS]() {
         for (int i = 0; i < TOTAL_ITEMS; ++i) {
             buffer->read([i, &target](uint8_t* buffer){
                 std::memcpy(&target[i][0], buffer, ITEM_SIZE);
@@ -136,7 +136,7 @@ TEST_F(ThreadSafeRingBufferTest, ReadWriteToBufferWithOverwrite) {
     EXPECT_TRUE(buffer->empty());
     EXPECT_FALSE(buffer->full());
 
-    std::thread producer([this, &source]() {
+    std::thread producer([this, &source, TOTAL_ITEMS]() {
         for (int i = 0; i < TOTAL_ITEMS; ++i) {
             buffer->write_overwrite([i, &source](uint8_t* buffer){
                 std::memcpy(buffer, &source[i][0], ITEM_SIZE);
@@ -148,7 +148,7 @@ TEST_F(ThreadSafeRingBufferTest, ReadWriteToBufferWithOverwrite) {
     // allowing sufficient time for the producer thread to be
     // completely done
     std::this_thread::sleep_for(1s);
-    std::thread consumer([this, &target]() {
+    std::thread consumer([this, &target, TOTAL_ITEMS]() {
         for (int i = 0; i < TOTAL_ITEMS; ++i) {
             buffer->read_timeout([i, &target](uint8_t* buffer){
                 if (buffer != nullptr)
```

headlee, 2023-12-06:
@Samahu Any updates on getting a Windows build tested? I've been able to get to the point where the os_driver is transitioning to the activate stage but then I crash:

```
[os_driver.EXE-1] [INFO] [1701874177.329727900] [ouster_driver]: reset service created
[os_driver.EXE-1] [INFO] [1701874177.336020600] [ouster_driver]: get_metadata service created
[os_driver.EXE-1] [INFO] [1701874177.340867300] [ouster_driver]: get_config service created
[os_driver.EXE-1] [INFO] [1701874177.341119800] [ouster_driver]: set_config service created
[INFO] [launch.user]: os_driver activating...
[ERROR] [os_driver.EXE-1]: process has died [pid 26160, exit code 3221225477, cmd 'C:\Users\jon\source\repos\ouster-ros\install\lib\ouster_ros\os_driver.EXE --ros-args -r __node:=ouster_driver -r __ns:=/ --params-file ouster-ros\config\community_driver_config.yaml'].
```

Still trying to figure out how to debug this properly on Windows - I'll let you know if I figure anything else out.

Samahu, 2023-12-06:
I spent little time to test the driver on Windows but then got busy with other higher priority tasks. Try adding debug launch prefix `prefix=['xterm -e gdb -ex run --args']` and see if that provide some more details on what the issue might be.